tensorflow/tflite-micro #438: TCN models containing expand dims and conv1d layers induce "Arena size is too small issue"

swapnilsayansaha, 2021-08-19:
I am trying to upload some lightweight TCN models to STM32L476RG Nucleo running Mbed OS.

Example model 1 (the tcn layer is from keras tcn library https://github.com/philipperemy/keras-tcn):
```
Model: "model"
__________________________________________________________________________________________________
Layer (type)                    Output Shape         Param #     Connected to                     
==================================================================================================
input_1 (InputLayer)            [(None, 400, 6)]     0                                            
__________________________________________________________________________________________________
tcn (TCN)                       (None, 32)           31840       input_1[0][0]                    
__________________________________________________________________________________________________
tf.reshape (TFOpLambda)         (None, 32, 1)        0           tcn[0][0]                        
__________________________________________________________________________________________________
max_pooling1d (MaxPooling1D)    (None, 16, 1)        0           tf.reshape[0][0]                 
__________________________________________________________________________________________________
flatten (Flatten)               (None, 16)           0           max_pooling1d[0][0]              
__________________________________________________________________________________________________
pre (Dense)                     (None, 32)           544         flatten[0][0]                    
__________________________________________________________________________________________________
velx (Dense)                    (None, 1)            33          pre[0][0]                        
__________________________________________________________________________________________________
vely (Dense)                    (None, 1)            33          pre[0][0]                        
==================================================================================================
Total params: 32,450
Trainable params: 32,450
Non-trainable params: 0
```
The above model consumes 162 kB of flash (1 MB capacity) and a total of ~50 kB (128 kB capacity) of RAM on the Mbed board (includes other dependencies) for arena of size 32X1000.

Example model 2 (which is even lighter!):
```
__________________________________________________________________________________________________
Layer (type)                    Output Shape         Param #     Connected to                     
==================================================================================================
input_2 (InputLayer)            [(None, 140, 1)]     0                                            
__________________________________________________________________________________________________
conv1d (Conv1D)                 (None, 140, 2)       22          input_2[0][0]                    
__________________________________________________________________________________________________
batch_normalization (BatchNorma (None, 140, 2)       8           conv1d[0][0]                     
__________________________________________________________________________________________________
zero_padding1d (ZeroPadding1D)  (None, 140, 2)       0           batch_normalization[0][0]        
__________________________________________________________________________________________________
conv1d_1 (Conv1D)               (None, 140, 11)      242         zero_padding1d[0][0]             
__________________________________________________________________________________________________
dropout (Dropout)               (None, 140, 11)      0           conv1d_1[0][0]                   
__________________________________________________________________________________________________
zero_padding1d_1 (ZeroPadding1D (None, 140, 11)      0           dropout[0][0]                    
__________________________________________________________________________________________________
conv1d_2 (Conv1D)               (None, 140, 11)      1331        zero_padding1d_1[0][0]           
__________________________________________________________________________________________________
batch_normalization_1 (BatchNor (None, 140, 11)      44          conv1d_2[0][0]                   
__________________________________________________________________________________________________
conv1d_3 (Conv1D)               (None, 140, 11)      22          batch_normalization[0][0]        
__________________________________________________________________________________________________
dropout_1 (Dropout)             (None, 140, 11)      0           batch_normalization_1[0][0]      
__________________________________________________________________________________________________
batch_normalization_2 (BatchNor (None, 140, 11)      44          conv1d_3[0][0]                   
__________________________________________________________________________________________________
add (Add)                       (None, 140, 11)      0           dropout_1[0][0]                  
                                                                 batch_normalization_2[0][0]      
__________________________________________________________________________________________________
activation (Activation)         (None, 140, 11)      0           add[0][0]                        
__________________________________________________________________________________________________
zero_padding1d_2 (ZeroPadding1D (None, 140, 11)      0           activation[0][0]                 
__________________________________________________________________________________________________
conv1d_4 (Conv1D)               (None, 140, 11)      1331        zero_padding1d_2[0][0]           
__________________________________________________________________________________________________
batch_normalization_3 (BatchNor (None, 140, 11)      44          conv1d_4[0][0]                   
__________________________________________________________________________________________________
[... 208 lines cut ...]
            }
            t.stop();
            printf("Latency: %f\n", t.read());
            t.reset();
            float out1 = output->data.f[0];
            float out2 = output->data.f[1];
            printf("Output: %f, %f",out1, out2);
            //thread_sleep_for(50);
        }
    } else {
        while(1){
            printf("Fatal Error\n");
        }
 
    }
}

```
The code compiles without any error. However, very surprisingly, during runtime, I get the following error for both models!

``` Arena size is too small for all buffers. Needed Arena size is too small for all buffers. Needed 4294960256 but only X was available. AllocateTensors() failed ``` (X varies, depending on arena size).

This is surely some bug in tensorflow, because the 2nd model was shown to run on the exact same hardware via TF Lite Micro: https://github.com/pulp-platform/ecg-tcn. 

TF 2.4 does not find expand dims, while TF 2.5 and 2.6 (this repo) complains about the arena size being unrealistic.

advaitjain, 2021-08-19:
Here are some debugging tips that may be useful:

 * Start by ensuring that you can use the model with TFLM on x86 and then attempting to run inference on your target platform.

 * To that end, create a benchmark for your model, similar to https://github.com/tensorflow/tflite-micro/blob/169a27b2048fdfc58ad815f22de6267e4ea8e86a/tensorflow/lite/micro/benchmarks/person_detection_benchmark.cc so that the errors can be reproduced more easily.

 * On x86, you can make the arena size as large as needed, and then use the recording memory APIs to get more insight into what is happening for your specific model: https://github.com/tensorflow/tflite-micro/blob/169a27b2048fdfc58ad815f22de6267e4ea8e86a/tensorflow/lite/micro/docs/memory_management.md#recording-memory-apis

You may also want to raise an issue on https://github.com/pulp-platform/ecg-tcn to find out more details about their use of TFLM.

closed 2021-08-19 by advaitjain

swapnilsayansaha, 2021-08-19:
Some more information (some of which is relevant to running the model on x86):

When I try to load an interpreter object on x86 using tf.lite.Interpreter(args.tflite_model), I get the following error:
```
tensorflow/lite/core/subgraph.cc BytesRequired number of elements overflowed.
Tensor 90 is invalidly specified in schema.
tensorflow/lite/core/subgraph.cc BytesRequired number of elements overflowed.
Tensor 97 is invalidly specified in schema.
tensorflow/lite/core/subgraph.cc BytesRequired number of elements overflowed.
Tensor 106 is invalidly specified in schema.
tensorflow/lite/core/subgraph.cc BytesRequired number of elements overflowed.
Tensor 113 is invalidly specified in schema.
tensorflow/lite/core/subgraph.cc BytesRequired number of elements overflowed.
Tensor 122 is invalidly specified in schema.
tensorflow/lite/core/subgraph.cc BytesRequired number of elements overflowed.
Tensor 129 is invalidly specified in schema.
tensorflow/lite/core/subgraph.cc BytesRequired number of elements overflowed.
Tensor 138 is invalidly specified in schema.
tensorflow/lite/core/subgraph.cc BytesRequired number of elements overflowed.
Tensor 145 is invalidly specified in schema.
tensorflow/lite/core/subgraph.cc BytesRequired number of elements overflowed.
Tensor 154 is invalidly specified in schema.
tensorflow/lite/core/subgraph.cc BytesRequired number of elements overflowed.
Tensor 161 is invalidly specified in schema.
tensorflow/lite/core/subgraph.cc BytesRequired number of elements overflowed.
Tensor 170 is invalidly specified in schema.
tensorflow/lite/core/subgraph.cc BytesRequired number of elements overflowed.
Tensor 177 is invalidly specified in schema.
tensorflow/lite/core/subgraph.cc BytesRequired number of elements overflowed.
Tensor 186 is invalidly specified in schema.
tensorflow/lite/core/subgraph.cc BytesRequired number of elements overflowed.
Tensor 193 is invalidly specified in schema.
```
Possible solutions are given here, pointing to problems with input shape on CNNs: https://stackoverflow.com/questions/64882214/runtimeerror-when-calling-allocate-tensors-on-converted-tflite-model and https://stackoverflow.com/questions/63500096/tensorflow-lite-core-subgraph-cc-bytesrequired-number-of-elements-overflowed-no

However, I can't use the solution as I the interpreter won't even load the tflite model, neither am I using built in TF CNNs

swapnilsayansaha, 2021-08-19:
Attached is the tflite model: https://drive.google.com/file/d/1swiQmoH_Rxy-f422eYnrVsDzIwpn-W7N/view?usp=sharing

swapnilsayansaha, 2021-08-19:
I think I solved the problem. It's got to do with older versions of TF (TF 2.4) not supporting dilation rates greater than 1. I swtiched the training version of TF to 2.5 and everything seems to be working fine.

martingcavallo, 2021-09-13:
Hi @swapnilsayansaha !

Something that I want to ask you if if you could convert the second model from (https://github.com/pulp-platform/ecg-tcn) using **Conv1D** or did you converted the equivalent model using Conv2D as explain in (https://github.com/pulp-platform/ecg-tcn).

I undestand that Conv1D is not still supported in tflite micro, isn't it?

Regards,
Martin

swapnilsayansaha, 2021-09-13:
@martingcavallo  https://github.com/tensorflow/tensorflow/issues/39823 https://github.com/tensorflow/tensorflow/pull/28410 

It was a bug with dilated conv support. It got fixed in recent versions of TF. I was able to convert my model to TFLM format without issues in TF 2.5++.

martingcavallo, 2021-09-18:
Ok. So I undesrtand that Convolutional 1D layer is supported by tflite micro, is that right??

On the other hand, I want to deploy a model on an arm mbed enable board too. But I do not know how to follow after that I execute the following command:

`make -f tensorflow/lite/micro/tools/make/Makefile TARGET=cortex_m_generic TARGET_ARCH=cortex-m4 OPTIMIZED_KERNEL_DIR=cmsis_nn microlite`

Since June, when tflite-micro was moved to this repository, seems like the process to deploy changed a bit, and some tutorials like the ones that are in O'Reilly TinyML book seems to be obsolete or deprecated. Because I do not find the directories that are mentioned in these tutorials after executing make command.

Did you use the same make command? Did you use Mbed Studio?? How to integrate tflite-micro to Mbed Studio??

Sorry for a lot of quesetions.

Regards,
Martin

swapnilsayansaha, 2021-09-18:
https://www.youtube.com/watch?v=gDFWCxrJruQ

Check this out (although its not for Mbed but concepts are similar). Basically you write your own main.cpp file based on the examples given in TFLM (e.g. the hello world main.cpp file). You then copy all the necessary dependencies from your make folder as shown in the video to your Mbed project directory. And yes you can use Mbed Studio to do it. 

Also I Used TF2.5 instead of 2.6 to compile directly for Mbed targets (target arch = mbed) They removed Mbed target in TF2.6.