tensorflow/tflite-micro #449: warning: 'resolved_axis' may be used uninitialized in this function (reduce.cc)

tcal-x, 2021-08-20:
This was seen when we attempted to bump the tflite-micro submodule in our project up to current main, from about a month ago (https://github.com/google/CFU-Playground/issues/230#issuecomment-900592633).

Since we compile TFL-micro with `-Werror` (using riscv gcc v8.3), this warning becomes an error for us:
```
In file included from /home/tim/tcal-x/CFU-Playground/proj/proj_template_v/build/src/tensorflow/lite/kernels/internal/common.h:29,
                 from /home/tim/tcal-x/CFU-Playground/proj/proj_template_v/build/src/tensorflow/lite/kernels/internal/reference/reduce.h:19,
                 from /home/tim/tcal-x/CFU-Playground/proj/proj_template_v/build/src/tensorflow/lite/micro/kernels/reduce.cc:16:
/home/tim/tcal-x/CFU-Playground/proj/proj_template_v/build/src/tensorflow/lite/kernels/internal/types.h: In function 'TfLiteStatus tflite::ops::micro::reduce::EvalMean(TfLiteContext*, TfLiteNode*)':
/home/tim/tcal-x/CFU-Playground/proj/proj_template_v/build/src/tensorflow/lite/kernels/internal/types.h:185:33: warning: 'resolved_axis' may be used uninitialized in this function [-Wmaybe-uninitialized]
         if (idx == axis[axis_idx]) {
                    ~~~~~~~~~~~~~^
/home/tim/tcal-x/CFU-Playground/proj/proj_template_v/build/src/tensorflow/lite/kernels/internal/types.h:185:33: warning: 'resolved_axis' may be used uninitialized in this function [-Wmaybe-uninitialized]
         if (idx == axis[axis_idx]) {
                    ~~~~~~~~~~~~~^
/home/tim/tcal-x/CFU-Playground/proj/proj_template_v/build/src/tensorflow/lite/kernels/internal/types.h:185:33: warning: 'resolved_axis' may be used uninitialized in this function [-Wmaybe-uninitialized]
         if (idx == axis[axis_idx]) {
                    ~~~~~~~~~~~~~^
```

I think it's referencing this: https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/kernels/reduce.cc#L149 , and I can't see where/how it gets initialized.

If you *don't* see these warnings, then there must be something wrong in our usage of tfl-micro.

tcal-x, 2021-08-20:
When I compiled without `-Werror`, there were no additional warnings and compilation finished successfully.

advaitjain, 2021-08-20:
As an aside, if you'd like to avoid pulling in a whole lot of unrelated code into your repo, you could use TFLM's new support for external integration:
 * https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/docs/new_platform_support.md
 * https://github.com/tensorflow/tflite-micro/issues/407

To address this specific issue, while the code in question is certainly complex, AFAICT this is a false positive.

The variable is passed as a pre-allocated buffer and gives the impression of being an input parameter. However, if we trace the call stack it is initialized.

For example,
https://github.com/tensorflow/tflite-micro/blob/a9f2e03b943990bc958ce3b92ca76ffc27fcf6d6/tensorflow/lite/micro/kernels/reduce.cc#L171-L177

calls into
https://github.com/tensorflow/tflite-micro/blob/a9f2e03b943990bc958ce3b92ca76ffc27fcf6d6/tensorflow/lite/kernels/internal/reference/reduce.h#L215-L240

which then initializes the variable in
https://github.com/tensorflow/tflite-micro/blob/a9f2e03b943990bc958ce3b92ca76ffc27fcf6d6/tensorflow/lite/kernels/internal/reference/reduce.h#L107-L109

It may be worth adding `-Wno-maybe-uninitialized` to the flags for TFLM in your repo.

For completeness, here are the flags that the TFLM repo uses:

gcc on x86:
https://github.com/tensorflow/tflite-micro/blob/a9f2e03b943990bc958ce3b92ca76ffc27fcf6d6/tensorflow/lite/micro/tools/make/Makefile#L130-L145

clang on x86:
https://github.com/tensorflow/tflite-micro/blob/a9f2e03b943990bc958ce3b92ca76ffc27fcf6d6/tensorflow/lite/micro/build_def.bzl#L1-L6

tcal-x, 2021-08-20:
Thank you for the detailed answer, @advaitjain .   We are currently manually copying the directories and files that we need from TFLM; we should switch over to the new support.   

FYI @alanvgreen

closed 2021-08-20 by tcal-x