tensorflow/tflite-micro #455: Tensor data is incorrectly loaded from FlatBuffers on big-endian targets

mateusz-holenko, 2021-08-23:
Currently `InitializeTfLiteTensorFromFlatbuffer` and `InitializeTfLiteEvalTensorFromFlatbuffer` refer to a data in a FlatBuffer using a direct pointer - an example is:

 https://github.com/tensorflow/tflite-micro/blob/a9f2e03b943990bc958ce3b92ca76ffc27fcf6d6/tensorflow/lite/micro/micro_allocator.cc#L440

Because data in FlatBuffers is always little-endian, on big-endian machines it causes problems because any type wider than 8 bits will be read with bytes swapped.

advaitjain, 2021-08-23:
Cross-referencing https://github.com/tensorflow/tflite-micro/issues/304 and https://github.com/tensorflow/tflite-micro/pull/456#issuecomment-903921078

advaitjain, 2021-11-17:
With https://github.com/tensorflow/tflite-micro/pull/689, we are removing endian changes at runtime. Instead, we would like for the byte-swapping to be external to the interpreter.

closed 2021-11-17 by advaitjain