tensorflow/tflite-micro #457: Add uint8 test cases back for the detection post-process kernel

advaitjain, 2021-08-23:
With https://github.com/tensorflow/tflite-micro/pull/428 all uint8 support was removed from the TFLM kernels. However, detection_postprocess is a special case where we likely still need to support uint8.

Tagging @mansnils and @freddan80 for confirmation.

mansnils, 2021-08-24:
@advaitjain We don't have any requirements for uint8 for this kernel. We might previously had a uint8 model that needed this but not anymore. So no changes needed, at least not from our side.

advaitjain, 2021-08-24:
ok, thanks for the information. I'm going to close this bug without any changes then since our strong preference is to not have any uint8 support.

closed 2021-08-24 by advaitjain

advaitjain, 2021-08-24:
Cross-referencing an internal thread on this topic: https://chat.google.com/room/AAAAOhoX8KA/HqMY9uJ6RJc