tensorflow/tflite-micro #458: ESP32 project generation broken by data file changes

petewarden, 2021-08-23:
The ESP32 community build (for example https://github.com/tensorflow/tflite-micro/runs/3399037264?check_suite_focus=true) is currently broken, and the cause appears to be the recent changes to the data file code generation. 

```
../main/main_functions.cc:20:10: fatal error: hello_world_model_data.h: No such file or directory
 #include "hello_world_model_data.h"
```

ytwanghaoyu, 2021-09-02:
same issue, plz fix

ytwanghaoyu, 2021-09-02:
same issue. even generate our own hello_world_model_data.h file won't make it

advaitjain, 2021-09-07:
Tagging @mocleiri and @vikramdattu.

We're happy to receive any fixes for this issue, but please note that the recommended path forward is to switch ESP32 to the approach described in https://github.com/tensorflow/tflite-micro/issues/407

njeffrie, 2021-09-08:
We are moving away from makefile-based project generation. Please try the following:
```
python3 tensorflow/lite/micro/tools/project_generation/create_tflm_tree.py  --examples hello_world <output_dir> 
```

mocleiri, 2021-09-09:
I wasn't sure where the new repo's were going to be placed.  However it looks like they are all being created still within the tensorflow org but as separate repos prefixed with tflite-micro- name.

I'll see if I can fix this as part of breaking out a separate repo

advaitjain, 2021-09-09:
The ESP32 repo should be hosted external to the tensorflow org. @vikramdattu might be able to help here.

The Arduino repo is hosted on the tensorflow org because the TFLM team will be maintaining that repo.

mocleiri, 2021-09-26:
@vikramdattu are you able to create a tflite-micro-esp-idf-examples repo at https://github.com/espressif?

The current examples are for esp32 but I think more of the esp-idf supported chips could be run the tflite-micro examples.

esp32, esp32c3, esp32s2, esp32s3 especially those that support spi ram for larger models like person detection.

advaitjain, 2021-09-27:
I have manually disabled the ESP32 workflow (https://github.com/tensorflow/tflite-micro/actions/workflows/esp32.yml) since it has been broken for over a month.

vikramdattu, 2021-09-28:
@advaitjain @mocleiri 
Bit busy right now and as I do not have any estimate on time needed to complete the activity, not taken yet.
Will inform once I'm on it.