tensorflow/tflite-micro #464: Error while generating the project.

aditi230, 2021-08-24:
error:-find: ‘../google/’: No such file or directory
tensorflow/lite/micro/tools/make/Makefile:453: * Something went wrong with the flatbuffers download: The top-level downloads directory: tensorflow/lite/micro/tools/make/downloads does not exist.. Stop.

This is the error i faced by running the command
make -f tensorflow/lite/micro/tools/make/Makefile TARGET=esp generate_hello_world_esp_project

How should I resolve it?

njeffrie, 2021-09-03:
Try to run
```
make -f tensorflow/lite/micro/tools/make/Makefile TARGET=esp clean_downloads
make -f tensorflow/lite/micro/tools/make/Makefile TARGET=esp third_party_downloads
```

njeffrie, 2021-09-09:
Please note that makefile-based project generation will no longer work for some examples. Try instead to run:
```
python3 tensorflow/lite/micro/tools/project_generation/create_tflm_tree.py  --makefile_options TARGET=esp --examples hello_world <output_dir> 
```

closed 2021-09-09 by njeffrie