tensorflow/tflite-micro #475: micro_speech example build fails with Mbed

annietllnd, 2021-08-30:
Steps to reproduce
```
$ make -f tensorflow/lite/micro/tools/make/Makefile OPTIMIZED_KERNEL_DIR=cmsis_nn generate_micro_speech_mbed_project
$ cd tensorflow/lite/micro/tools/make/gen/linux_x86_64_default/prj/micro_speech/mbed/
$ mbed new .
$ mbed compile -m DISCO_F746NG -t GCC_ARM
```
Since patch #429 fails with following output:
```
Compile [ 56.3%]: main_functions.cc
[Fatal Error] main_functions.cc@22,10: tensorflow/lite/micro/examples/micro_speech/micro_speech_model_data.h: No such file or directory
[ERROR] ./tensorflow/lite/micro/examples/micro_speech/main_functions.cc:22:10: fatal error: tensorflow/lite/micro/examples/micro_speech/micro_speech_model_data.h: No such file or directory
22 | #include "tensorflow/lite/micro/examples/micro_speech/micro_speech_model_data.h"
   |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

The referenced file seems  to be missing in `tensorflow/lite/micro/examples/micro_speech/`. Possibly related to #423 #458

ytwanghaoyu, 2021-09-02:
Same issue, can't find the .h file

njeffrie, 2021-09-03:
I see... this was an unintended side-effect of my recent change. In the short term, if you `cp tensorflow/lite/micro/tools/make/gen/linux_x86_64_default/genfiles/tensorflow/lite/micro/examples/micro_speech/micro_speech_model_data.* tensorflow/lite/micro/tools/make/gen/linux_x86_64_default/prj/micro_speech/mbed/tensorflow/lite/micro/examples/micro_speech/` it should prevent the missing file error.

We are switching from Makefile-based project generation to the python script `tensorflow/lite/micro/tools/project_generation/create_tflm_tree.py` so you could try using this script instead.

annietllnd, 2021-09-03:
@mansnils fyi

mansnils, 2021-09-03:
As there is a workaround I think we can close this.

closed 2021-09-03 by mansnils

njeffrie, 2021-09-08:
I was successfully able to use create_tflm_tree.py as follows:
```
python3 tensorflow/lite/micro/tools/project_generation/create_tflm_tree.py  --makefile_options OPTIMIZED_KERNEL_DIR=cmsis_nn --examples micro_speech <output_dir> 
cd <output_dir>
mbed new .
mbed compile -m DISCO_F746NG -t GCC_ARM -D CMSIS_NN

Note: I had to:
cd mbed-os && mbed update master
```