tensorflow/tflite-micro #483: new tf-lite micro integration issues

TI-AviBerko, 2021-09-04:
@advaitjain 

Some more questions:
1.	I’m building with TF_LITE_STATIC_MEMORY, and I’m getting this compiler error about a private destructor: 'static void tflite::MicroErrorReporter::operator delete(void*)' is private within this context’ – I can get this to build by commenting out line 44 in micro_error_reporter.c
2.	What happened to “debug_log.cc”, how do I provide a debug reporter?
3.	We used to pass “TAGS=portable_optimized” to the old makefile to get an optimized depthwise x8 kernel, is this implementation still available?

advaitjain, 2021-09-09:
> @advaitjain
>
> Some more questions:
>
> 1. I’m building with TF_LITE_STATIC_MEMORY, and I’m getting this compiler error about a private destructor: 'static void tflite::MicroErrorReporter::operator [...]

I'm assumuing that you are talking about this line
https://github.com/tensorflow/tflite-micro/blob/5e65f8204588f0f7ae3e8cd36ac8c311f242f50e/tensorflow/lite/micro/micro_error_reporter.h#L43-L44

Can you share more information about your compiler version and if possible what build flags you are using?

It is ok to move `TF_LITE_REMOVE_VIRTUAL_DELETE` to public, similar to
https://github.com/tensorflow/tflite-micro/blob/5e65f8204588f0f7ae3e8cd36ac8c311f242f50e/tensorflow/lite/micro/micro_mutable_op_resolver.h#L39-L41

Please feel free to send a PR for that, if that works for you.

> 2. What happened to “debug_log.cc”, how do I provide a debug reporter?

The platform specific repo should provide the implementation of debug_log.cc.

Here is the example for sparkfun_edge:
https://github.com/advaitjain/tflite-micro-sparkfun-edge-examples/tree/84194626c20495df38981f3c5bcaeabd890a540c/tensorflow/lite/micro/sparkfun_edge

and Arduino:
https://github.com/tensorflow/tflite-micro-arduino-examples/blob/25e4ffffd86bc936f34a65946c3f52648d2098b9/src/tensorflow/lite/micro/system_setup.cpp

> 3. We used to pass “TAGS=portable_optimized” to the old makefile to get an optimized depthwise x8 kernel, is this implementation still available?

We have removed `portable_optimized` in favor of using the optimized kernels (such as `OPTIMIZED_KERNEL_DIR=cmsis_nn`, or `OPTIMIZED_KERNEL_DIR=xtensa`. Here is some additional context on this topic: https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/docs/optimized_kernel_implementations.md#why-not-optimize-the-portable-reference-kernels

Assuming that you are looking for optimizations for cortex-m, you invocation of the create_tflm_tree.py could look something like:
https://github.com/advaitjain/tflite-micro-sparkfun-edge-examples/blob/84194626c20495df38981f3c5bcaeabd890a540c/scripts/sync_from_tflite_micro.sh#L33-L42

TI-AviBerko, 2021-09-14:
> I'm assumuing that you are talking about this line
> 
> tflite-micro/tensorflow/lite/micro/micro_error_reporter.h
> 
> Lines 43 to 44 in 5e65f82
>  private: 
>   TF_LITE_REMOVE_VIRTUAL_DELETE 
> 

 - Indeed

Same issue at: tensorflow/lite/micro/memory_planner/greedy_memory_planner.h @ line 162

The gcc version we're using is: arm-none-eabi-gcc 7.2.1 20170904
flags:  -mcpu=cortex-m4 -march=armv7e-m -mthumb -mfloat-abi=hard -mfpu=fpv4-sp-d16 -DNDEBUG -DTF_LITE_STATIC_MEMORY -DCMSIS_NN -O3 -ffunction-sections -fdata-sections -g -gdwarf-3 -gstrict-dwarf -Wall -flto -fno-strict-aliasing

The CMSIS-NN kernels work as expected, thanks.

I have another question about the SYNC/CI badges: the SYNC action would obviously work fine for us, but we would prefer running the CI on our internal CI,  which cannot normally be triggered by github.com, do you know of a way we can have a CI badge by committing some kind of build proof/status (after polling the github repo)?

advaitjain, 2021-09-27:
Apologies for the delayed response. Please feel free to send a PR that moves `TF_LITE_REMOVE_VIRTUAL_DELETE` to be public (assuming that that fixes your build).

If your internal CI has a way to publish a badge to a public location, then we can certainly link to that from GitHub. For example, the [TF community supported build page](https://github.com/tensorflow/build#community-supported-tensorflow-builds) has a lot of build badges generated via Jenkins.

TI-AviBerko, 2021-10-03:
I went through all the flags, and it seems that passing -fno-rtti & -fno-exception resolves the virtual delete issue, so no need to for any PRs.

For what I understand, the badge is just and SVG file, I wonder if publishing it to a public location as you suggested is enough? we won't be able to provide a nice link to the jenkins run, unless no logs or just a single execution log is enough.

Regardless, I have our repo up and running: https://github.com/TexasInstruments/tensorflow-lite-micro-examples
I introduced a "targets" directory since we'll be supporting multiple targets (there are only 2 in there right now)

Thanks.

advaitjain, 2021-10-04:
That's great to hear!

We'll keep `-fno-rtti -fno-exceptions` if any other integrations run into the same issue.

Yes, publishing an svg to a public location is sufficient. It is up to you as the examples repo maintainer what type additional information you'd like to provide.

Please do send a PR adding your repository to the [Community Supported TFLM Examples](https://github.com/tensorflow/tflite-micro#community-supported-tflm-examples) table.

TI-AviBerko, 2021-10-04:
Regarding the support TFLM example table, our repo contains more than one platform, how shall we handle this?

I thought about having a CI badge per platform, does that make sense? how should we even display that in the table?

advaitjain, 2021-10-05:
I would recommend a single badge call it "TI dev boards" or something else that you feel is both descriptive and generic.

And clicking on that badge will go to your github repo's readme, where you can have a more detailed table. Something like: https://github.com/advaitjain/tflite-micro/blob/d28bba16c9d50d80fd646ac3b457446bb2f2a040/tensorflow/lite/micro/docs/local_continuous_builds/xtensa.md#summary

closed 2023-07-05 by github-actions[bot]

HCzou, 2024-05-17:
modify related "private" to "public", solved this issue.