tensorflow/tflite-micro #485: Arduino trained model initialization issues

bradinglis, 2021-09-07:
Hi,

I am having issues with the Arduino C++ library, specifically when using model data from a trained model using the scripts provided in `tensorflow/lite/micro/examples/magic_wand/train/`. I have provided details on how to reproduce the errors, using 2 different versions of the Arduino library (unsure which one is more useful for someone who knows what's going on).

**Platform:** Arduino Nano 33 BLE
Training the model using the scripts in `tensorflow/lite/micro/examples/magic_wand/train/` with the provided data, following the basic instructions in the README there.

Using both methods of installing the Arduino library, the examples were run as-is from the library and worked as expected. Only when replacing the model data with the newly trained model do errors occur.

### **When installing the library from the Arduino library manager**

Error through serial is reported:

```
Invoke() called after initialization failed

Invoke failed on index
```

This occurs when the model is invoked in the while loop in this project, however obviously there are issues initialising the model while in the `setup()` function. (Perhaps a seperate issue, but the error reporting functionality does not seem to work in this setup function, maybe an Arduino quirk.) 

### **When installing the library by cloning from `tensorflow/tflite-micro-arduino-examples` repository**

After uploading the sketch, the serial USB port of the Arduino is no longer detected. The only way to be able to access the port again is by double pressing reset (which I believe is bootloader mode). Very strange behaviour which I am not quite sure of the cause, although it is presumably due to the newly trained model.

### **My guess**

I can only assume that the supplied training scripts/Tensorflow python library has changed in such a way where they create a model which is unable to be initialised by the C++ micro library. I am unfortunately not knowledgable enough to dig any deeper, but I am fairly certain that this is not something just on my end. My apologies if there is a lack of clarity, let me know if more information is needed.

advaitjain, 2021-09-27:
Tagging @petewarden to answer Arduino-related questions.

petewarden, 2021-10-04:
Just a note to say that I'm looking into general Arduino issues this week and hope to have an update on this problem.

midiclorian, 2022-10-07:
Hello, is there a solution to this issue? I am having the same problem and I was curious if there is a workaround.
Thank you.

himikewu, 2023-02-18:
hi, I recently came across the same issue in my application. Then I compared the sample model and the newly trained model。I found the new model have a new layer named reshape .
So I add a new line code 
 micro_op_resolver.AddReshape();
Or you can just use 
 tflite::AllOpsResolver resolver;
to use the Allopsresplver
Both approaches can solve the problem.

closed 2023-02-20 by ddavis-2015