tensorflow/tflite-micro #496: Didn't find op for builtin opcode 'SPACE_TO_BATCH_ND' version '2'. An older version of this builtin might be supported. Are you using an old TFLite binary with a newer model?

martingcavallo, 2021-09-13:
Hi!

I am trying to deploy a TF Lite model to an **Arduino Nano 33 BLE Sense**, and I get the following error:

> Didn't find op for builtin opcode 'SPACE_TO_BATCH_ND' version '2'. An older version of this builtin might be supported. Are you using an old TFLite binary with a newer model?

I have trained and converted the model using **python 3.8.8** and **tensorflow 2.6.0**. After that, I converted the model using Post Training Quantization, I also run some inferences in Juptyter Notebook using TF Lite Interpreter without any problems.

Now I want to deploy this model to my Arduino Nano 33 BLE Sense boards, but I get the error described above. I am using **Arduino IDE** and the library **Arduino_TensorFlowLite version 2.4.0-ALPHA.**

What I can see is that in [all_apps_resolver.h](https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/all_ops_resolver.cc) is implemented the function ``AddSpaceToBatchNd();``, however, if I look for this function in the file ''all_ops_resolver.cc'' included in **Arduino_TensorFlowLite version 2.4.0-ALPHA.**, I saw that this function is missed, I mean, this function is not implemented.

So, I understand that the problem is the version of my Arduino_TensorFlowLite libreary. Is there any version that inlude ``AddSpaceToBatchNd();`` or it is not still implemented and it does not has support for Arduino library?

Thanks in advance!!

Best regards,
Martin

advaitjain, 2021-09-27:
Tagging @petewarden to answer Arduino-related questions.

antoine-lebaud, 2021-10-04:
Hi @martingcavallo,
I got the same issue as you mentioned. I would like to reproduce the steps you did. Would you recommend me any good resource to generate a .zip Arduino library?