tensorflow/tflite-micro #498: Memory arena threshold test is failing for FVP target

mansnils, 2021-09-13:
There is a comment:

// NOTE: These values are measured on x86-64:
// TODO(b/158651472): Consider auditing these values on non-64 bit systems.

Could it be related to why the tests fail?

How to reproduce: 
First remove the filtering of the test here: https://github.com/tensorflow/tflite-micro/blob/d41f2011f3da86e15cea83b5623800552dd52ed7/tensorflow/lite/micro/tools/make/targets/cortex_m_corstone_300_makefile.inc#L199
make -f tensorflow/lite/micro/tools/make/Makefile TARGET=cortex_m_corstone_300 TARGET_ARCH=cortex-m55 test_memory_arena_threshold_test

advaitjain, 2021-09-13:
Tagging @tensorflow/micro

deqiangc, 2021-09-21:
Indeed this failure is due to platform difference. Ideally, we should convert all these magic number to expressions that can change automatically across the platform, but that is going to take some time. So for now, the best option is not to enable this test on non 64 bit platform

advaitjain, 2021-09-27:
I'm going to close this issue for now. Refactoring the memory_arena_threshold_test to make it generalize across multiple platforms is not something that we are working on.

closed 2021-09-27 by advaitjain