tensorflow/tflite-micro #504: Issue With project_generation.py

embeddetech, 2021-09-13:
I am running an automated process that pulls TFLM into an existing VIsual Studio project. It was working before (a few weeks ago), but am seeing it break now.

The sequence is as follows:

- The repo is cloned into a temp folder
- I run python tensorflow/lite/micro/tools/project_generation/create_tflm_tree.py, [destination folder]
- This eventually internally calls make -f "tensorflow\lite\micro\tools\make\Makefile" third_party_downloads, where an exception is thrown at line 225 of create_tflm_tree.py where make is run for this command.
- The Makefile loads and crashes in tensorflow/lite/micro/tools/generate_cc_arrays.py", line 26: from PIL import Image throws an error, ModuleNotFoundError: No module named 'PIL'. This python file appears to be called in tensorflow\lite\micro\tools\make\helper_functions.inc line 499

There are several articles about the PIL module not being found, and how to fix it by installing pillow. Does pillow need to be manually installed to replace PIL for tflite-micro?

advaitjain, 2021-09-13:
Yes, pillow needs to be manually installed. And you might need Wave as well.

Here is where we have updated the CI to install these two dependencies:
https://github.com/tensorflow/tflite-micro/blob/1ef3830fa693bc5ab745100e53d55da97b4f8c29/.github/workflows/ci.yml#L40-L41

See related issue: https://github.com/tensorflow/tflite-micro/issues/401

closed 2021-09-13 by advaitjain

embeddetech, 2021-09-14:
Thank you, @advaitjain! So if I understand correctly, that action will only run on a pull request when building with bazel on ubuntu, is that correct? I can add this to my automation process.

advaitjain, 2021-09-14:
Correct, I only linked to the workflow yml as an example of what you will need to do.

embeddetech, 2021-09-14:
Update: 
I am using msys64 for make, and attempting to manually install Pillow in this global environment, using mintty with the command "python3 -m pip install Pillow". This command works fine when running in DOS which uses a different Python interpreter, but using the interpreter in msys64 seems to have issues caused by pip trying to build Pillow on the fly in a Windows environment as explained in this link:

https://github.com/python-pillow/Pillow/issues/2860

I am going to try to get Pillow installed without being built, just thought you might like to be aware of this potential issue building on Windows.