tensorflow/tflite-micro #508: arm-gcc download is failing

advaitjain, 2021-09-15:
```
wget https://developer.arm.com/-/media/Files/downloads/gnu-rm/10-2020q4/gcc-arm-none-eabi-10-2020-q4-major-x86_64-linux.tar.bz2
```

is failing with:
```
--2021-09-15 11:38:06--  https://developer.arm.com/-/media/Files/downloads/gnu-rm/10-2020q4/gcc-arm-none-eabi-10-2020-q4-major-x86_64-linux.tar.bz2
Resolving developer.arm.com (developer.arm.com)... 104.93.139.97
Connecting to developer.arm.com (developer.arm.com)|104.93.139.97|:443... connected.
ERROR: The certificate of ‘developer.arm.com’ is not trusted.
ERROR: The certificate of ‘developer.arm.com’ doesn't have a known issuer.
```

The current workaround is to add `--no-check-certificate` to the wget command, but we should revert that once the underlying issue is fixed.

Tagging @mansnils @freddan80

advaitjain, 2021-09-15:
Internal bug: http://b/200052821

freddan80, 2021-09-16:
We'll have a look at this issue.

freddan80, 2021-09-16:
Fix is being worked on by our IT department. We hope hope to have a resolution soon.

freddan80, 2021-09-17:
Should be fixed now. It works on our end

PR #522 (2021-09-20): Revert "Workaround for failing arm-gcc download."
Reverts tensorflow/tflite-micro#509

ARM has fixed the issue at their end: https://github.com/tensorflow/tflite-micro/issues/508#issuecomment-921529046

BUG=Fixes #508
closed 2023-07-03 by mansnils