tensorflow/tflite-micro #521: Generate the examples for ESP32 fails on windows

MarcoLeti, 2021-09-20:
I am trying to build the model https://github.com/tensorflow/tflite-micro/tree/main/tensorflow/lite/micro/examples/person_detection into an ESP32 cam module.
Unfortunately it fails in the step "Generate the example":
```
Generate the examples
```
with the following error:
```
PS F:\git\tflite-micro> make -f tensorflow/lite/micro/tools/make/Makefile TARGET=esp generate_person_detection_esp_project                                                                                                                      process_begin: CreateProcess(NULL, uname -m, ...) failed.                                                               tensorflow/lite/micro/tools/make/Makefile:47: pipe: No error                                                            -m was unexpected at this time.                                                                                         FIND: Parameter format not correct                                                                                      FIND: Parameter format not correct                                                                                      File not found - "*.c"                                                                                                  Access denied - TENSORFLOW/LITE                                                                                         File not found - -TYPE                                                                                                  File not found - D                                                                                                      File not found - \(                                                                                                     File not found - -PATH                                                                                                  Access denied - TENSORFLOW/LITE/EXPERIMENTAL                                                                            File not found - -O                                                                                                     File not found - -PATH                                                                                                  Access denied - TENSORFLOW/LITE/MICRO                                                                                   File not found - \)                                                                                                     File not found - -PRUNE                                                                                                 File not found - -FALSE                                                                                                 File not found - -O                                                                                                     File not found - -NAME                                                                                                  The syntax of the command is incorrect.                                                                                 process_begin: CreateProcess(NULL, bash F:\git\tflite-micro\tensorflow\lite\micro\tools\make\flatbuffers_download.sh tensorflow/lite/micro/tools/make/downloads, ...) failed.                                                                   tensorflow/lite/micro/tools/make/Makefile:462: pipe: Bad file descriptor                                                tensorflow/lite/micro/tools/make/Makefile:464: *** Something went wrong with the flatbuffers download: .  Stop. 
```
I  am on windows. Is this a problem? I have installed make for windows.

Thanks

vikramdattu, 2021-09-21:
I think this is not specific to ESP32.
Can you please check without specifying TARGET, i.e, for host architecture?

MarcoLeti, 2021-09-21:
thanks @vikramdattu I tried
```
make -f tensorflow/lite/micro/tools/make/Makefile
```
but I got the same error...I know it is not connected to esp and it is just a generation of example files but it seems that it cannot be compiled on windows, I saw other people having the same problem, as a temporary workaround, do you think it is possible to 'make' the files in a linux machine and then bring everything on windows? Thanks...

vikramdattu, 2021-09-21:
@MarcoLeti that should work.

advaitjain, 2021-09-21:
Couple of things:

 * Per https://github.com/tensorflow/tflite-micro/issues/170#issuecomment-860888987
 > Windows is currently fully community supported and the builds are not part of any CI or something that the TFLM engineers are set up to test. Often errors are the result of some incompatibility with the shell commands (which appears to be the case here).
> 

 * https://github.com/tensorflow/tflite-micro/issues/458
    * the makefile based project generation for ESP32 is currently broken
    * our recommendation is to switch to the Python based project generation since that is more robust and has better support https://github.com/tensorflow/tflite-micro/issues/445 and https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/docs/new_platform_support.md

I'm going to close this issue since the makefile based project generation is not something we can support. We would still be open to reviewing PR with fixes for the time-being but would strongly recommend that ESP32 move in the direction outlined in https://github.com/tensorflow/tflite-micro/issues/407

closed 2021-09-21 by advaitjain