tensorflow/tflite-micro #527: No rule to make target 'tensorflow/lite/micro/tools/make/targets/arc_vpx_makefile.inc'

mohanish3795, 2021-09-22:
While building arc_mli, I am getting following error

Command given: make -f tensorflow/lite/micro/tools/make/Makefile TARGET=arc_vpx OPTIMIZED_KERNEL_DIR=arc_mli generate_person_detection_int8_make_project

Error : No rule to make target 'tensorflow/lite/micro/tools/make/targets/arc_vpx_makefile.inc' 

Looks like the arc_vpx_makefile.inc file is missing.

gerbauz, 2021-09-28:
Hi @mohanish3795, sorry for the delay.

First of all, I temporary suggest to use our fork: https://github.com/foss-for-synopsys-dwc-arc-processors/tflite-micro, upstream version may not work.
Also, to build example for VPX, please use following instructions:
https://github.com/foss-for-synopsys-dwc-arc-processors/tflite-micro/tree/fork_main/tensorflow/lite/micro/examples/person_detection#generate-example-project-1

Regards,
Artem

petewarden, 2021-10-04:
@gerbauz Let us know if you have a pull request pending to fix this in the main branch! I'll close this issue for now since there's a workaround.

closed 2021-10-04 by petewarden