tensorflow/tflite-micro #534: No makefile for Sparkfun Edge

ClaudeZo, 2021-09-23:
Dear Team,

It is the first time I am using the TFLM and the Sparkfun Edge board. I am following the tutorial on Sparkfun and the Colab
tutorial : https://codelabs.developers.google.com/codelabs/sparkfun-tensorflow/#2
I can compile and link an example for the Sparkfun Edge using the Ambiq SDK.
I am trying now to generate an example with TFLM that I want to run on the Sparkfun Edge to check if the workflow is properly set up.
I installed all the tools needed on Windows.

When I launch make, the sparkfun_edge_makefile.inc is not found because the makefile is not in the GitHub repo.

$ make -f tensorflow/lite/micro/tools/make/Makefile TARGET=sparkfun_edge micro_speech_bin
tensorflow/lite/micro/tools/make/downloads/flatbuffers already exists, skipping the download.
tensorflow/lite/micro/tools/make/downloads/pigweed already exists, skipping the download.
tensorflow/lite/micro/tools/make/Makefile:497: tensorflow/lite/micro/tools/make/targets/sparkfun_edge_makefile.inc: No such file or directory
make: *** No rule to make target 'tensorflow/lite/micro/tools/make/targets/sparkfun_edge_makefile.inc'.  Stop.

Any recommandation or alternative ?
Many thanks in advance for your kind help
ClaudeZo

advaitjain, 2021-09-23:
We are looking for someone to maintain the Sparkfun port of TFLM, see https://github.com/tensorflow/tflite-micro/issues/407#issuecomment-898660589 for additional context.

For now, you can try https://github.com/advaitjain/tflite-micro-sparkfun-edge-examples, with the caveat that we are not able to provide any support.

If you would be interested in being a maintainer for TFLM on Sparkfun Edge, then please let us know.

closed 2021-09-23 by advaitjain

magarita2022, 2023-01-27:
> We are looking for someone to maintain the Sparkfun port of TFLM, see [#407 (comment)](https://github.com/tensorflow/tflite-micro/issues/407#issuecomment-898660589) for additional context.
> 
> For now, you can try https://github.com/advaitjain/tflite-micro-sparkfun-edge-examples, with the caveat that we are not able to provide any support.
> 
> If you would be interested in being a maintainer for TFLM on Sparkfun Edge, then please let us know.

What is the make command to compile the micro-speech for sparkFun-edge, please?