tensorflow/tflite-micro #564: MagicWand example selftrained model

baenji456, 2021-10-04:
Dear tflite-micro team,

Me and my friend were working on getting the MagicWand example to work with our own data.

While the example would run, with the included model, it wouldn't work, because we have a different accelerometer. That's why we wanted to train our own model on our own data. The process of making the model worked perfectly. The MagicWand example also built and compiled without problems, but when we tried running it on our Microcontroller we got an error at line 87 in main_functions.cc.

 

>   if ((model_input->dims->size != 4) || (model_input->dims->data[0] != 1) ||
>       (model_input->dims->data[1] != 128) ||
>       (model_input->dims->data[2] != kChannelNumber) ||
>       (model_input->type != kTfLiteFloat32)) {
>     TF_LITE_REPORT_ERROR(error_reporter,
>                          "Bad input tensor parameters in model");
>     return;
>   }

When we tried training the model with your data from

> download.tensorflow.org/models/tflite/magic_wand/data.tar.gz

we got the same error, even when we tried building and training it in the [Google collab](https://colab.research.google.com/github/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/examples/magic_wand/train/train_magic_wand_model.ipynb), were they say:

> It will produce the same model used in the magic_wand example application.

We cant make out a difference between the models other than the size, which is:

- 20288 Bytes in our self trained models,
- 9760 Bytes in our selftrained models, when quantized
- and 19616 Bytes in the given model from the MagicWand example.

When we look up the values in model_input->dims, we can see, that the values that are checked in line 87 and following aren't correct, but we don't know at which part of the process the mistake happens.

Also: We tried making the .tflm file into a .cc file with the xxd command in Linux, as well as with our own Python Script. The output was the same.

Additional information:
- We are working in IAR RISC-V
- on a RISC-V based microcontroller

We would love to get some help on this issue.

Best regards
Max & Benjamin

petewarden, 2021-10-04:
Sorry you're hitting problems! Can you share a link to your model (privately with petewarden@google.com if necessary) and the error message you see?

baenji456, 2021-10-06:
[Attached ](https://github.com/baenji456/magicwand-models)you will find our TFLite models and the .cc model_data files, that we are including in the example. Also you will find the original model, which is included in the original example.
 
The Error we should get is “Bad input tensor parameters in model”, but the Debugger breaks at the first if statement in line 86 of main_functions.cc.
When we check the dimensions that are checked in the statement and are written in “model_input->dims” (see line 86-89 in main_functions.cc) we can see, that those dimensions are wrong, but we can’t make out why the dimensions of the model are wrong.
e.g.: (model_input->dims->size != 4)      model_input->dims->size should be 4 otherwise the Error “Bad input tensor parameters in model” occurs, but when we look at model_input->dims->size we see, that it is 8.