tensorflow/tflite-micro #581: How big is the flashed model on the hardware compared to the tf lite model?

RomanFoell, 2021-10-09:
Hi,

When I generate a C file from my tf lite model mit xxd, the file is nearly 10 times bigger than the lite model file. I also saw the big size differences in the zip download from one of your examples.

Does this mean the flash of the hardware has to be a big as the generate C file to run the code on hardware. I am trying to understand the examples first before I create my own model and implement it on the hardware, but this is a question which came into my mind. Why is the C file that much bigger than the tf lite model?

Thanks.

mocleiri, 2021-10-10:
Its not the C file that goes onto the board.  Its compiled first.  

The number of bytes should be the same between the size of the model.tflite file and the size listed within the C array file.

If you don't have a filesystem then you should use the model in the array approach but if you are learning and experimenting with different model's its possible to read the model.tflite into a byte array from a file system at runtime.

In both case the size of the model is the number of bytes of the model.tflite file.

RomanFoell, 2021-10-10:
Thanks for your answer.