tensorflow/tflite-micro #685: Fully connected node without bias has invalid parameters when run with xtensa nnlib on HiFi4

jgorham, 2021-11-15:
I am trying to deploy a model trained in tensorflow on the HiFi4 using TFLM. After training a simple NN in tensorflow (a one layer multilayer perception without a bias term) and converting to tflite using the standard optimizations, I can try to run the model on the HiFi4 and run inference. Everything appears to work fine until I call invoke on the interpreter, in which case the return code is -1.

After adding some more logging, I have noticed that the input to the fully connected nnlib node is 
```
EvalQuantizedInt8: batch=0 p_out=0x003ed630 p_weight=0x003d7b7c p_inp=0x003ecf30 p_bias=0x00000000
```
Hence `p_bias` is being set to null. However, this is unfortunately prohibited by the xtensa nnlib library:

https://github.com/foss-xtensa/nnlib-hifi4/blob/42da7e5c17844a0a83dc6b32f06258504bcdd373/xa_nnlib/algo/kernels/fc/hifi4/xa_nn_fully_connected.c#L53

If I retrain my tensorflow model but make sure to include a bias term, everything works fine. I believe a similar error was patched in the CMSIS library: https://github.com/ARM-software/CMSIS_5/issues/920. Would it be possible to add some similar protection here?

lukmanr-cadence, 2021-11-17:
@cad-audio , please look into this issue.

advaitjain, 2021-11-19:
Will be fixed with https://github.com/tensorflow/tflite-micro/pull/590 and https://github.com/tensorflow/tflite-micro/pull/711

PR #590 (2021-11-23): Handle nullptr bias in fully_connected op
* Validate tensor index in GetEvalInput.
* Check for nullptr bias tensor in fully connected kernel.

BUG=Fixes https://github.com/tensorflow/tflite-micro/issues/589, fixes #685  and http://b/205306469
closed 2021-11-23 by advaitjain