tensorflow/tflite-micro #707: Transpose conv kernel wrapper has bugs

keandersslab, 2021-11-19:
The Prepare stage treats the kernel params as if it was is of type TfLiteConvParams. It should be TfLiteTransposeConvParam*. We then get arbitary values for dilation and activation. This leads to errors when calling ComputePaddingHeightWidth() and PopulateConvolutionQuantizationParams().

This patch is shows a sufficient solution:
[patch.txt](https://github.com/tensorflow/tflite-micro/files/7569046/patch.txt)

petewarden, 2021-11-24:
Thanks @keandersslab ! I took a look at the patch and I see the issue. Would you be able to provide this as a pull request to make merging it easier? Having it as two separate PRs, one for this and another for #706 would help too.

Thanks again for your help improving the framework.

keandersslab, 2021-11-25:
> CAUTION: This email originated from outside of the organization. Do not click links or open attachments unless you recognize the sender and know the content is safe.

Sure Peter !

I doubt that I will have time right now to make test cases, but I will certainly be able provide pull requests by end of this week or early next week 😊

-Kjell Eirik Andersen

closed 2023-06-18 by github-actions[bot]