tensorflow/tflite-micro #716: model convert successfully, but invoke fail.

jiayue198509, 2021-11-23:
[Uploading tf-awake-1109crnn-81-int8-2.6.tflite.zip…]()
model convert successfully. but invoke fail.
CONV_2D exception: xa_nnlib_hifi5/algo/kernels/cnn/hifi5/xa_nn_matXvec_sym8sxasym8s_asym8s_circ.c

njeffrie, 2021-11-23:
Are you able to run this model using reference kernels (without OPTIMIZED_KERNEL_DIR=xtensa)?

jiayue198509, 2021-11-24:
> 

invoke still fail without OPTIMIZED_KERNEL_DIR=xtensa. The code block after runing some ops.

Node CONV_2D (number 1) invoke with status 0 
Node CONV_2D (number 2) invoke 
Node CONV_2D (number 2) invoke with status 0 
Node CONV_2D (number 3) invoke 
Node CONV_2D (number 3) invoke with status 0 
Node CONV_2D (number 4) invoke 
Node CONV_2D (number 4) invoke with status 0 
Node ADD (number 5) invoke 
Node ADD (number 5) invoke with status 0 
Node CONV_2D (number 6) invoke 
Node CONV_2D (number 6) invoke with status 0 
Node CONV_2D (number 7) invoke 
Node CONV_2D (number 7) invoke with status 0 
Node CONV_2D (number 8) invoke 
Node CONV_2D (number 8) invoke with status 0 
Node ADD (number 9) invoke 
Node ADD (number 9) invoke with status 0 
Node CONV_2D (number 10) invoke 
Node CONV_2D (number 10) invoke with status 0 
Node CONV_2D (number 11) invoke 
Node CONV_2D (number 11) invoke with status 0 
Node CONV_2D (number 12) invoke 
Node CONV_2D (number 12) invoke with status 0 
Node ADD (number 13) invoke 
Node ADD (number 13) invoke with status 0 
Node RESHAPE (number 14) invoke 
Node RESHAPE (number 14) invoke with status 0 
Node FULLY_CONNECTED (number 15) invoke 
[   80.658192@0]  audio_ddr_mngr: frddr stop success, fifo id 0, regbase:0x70, arb sts:0x10

njeffrie, 2021-11-24:
It seems like the link to the model above is broken - can you attach a new link to the model?

jiayue198509, 2021-11-24:
[tf-awake-1109crnn-81-int8-2.6.tflite.zip](https://github.com/tensorflow/tflite-micro/files/7592402/tf-awake-1109crnn-81-int8-2.6.tflite.zip)
The model link. Thanks.

njeffrie, 2021-11-24:
It looks like the model runs fine for me with reference kernels. Note that invoke status 0 means invoke completed with status kTfLiteOk. Did the logs get cut short or did FULLY_CONNECTED crash for you?

jiayue198509, 2021-11-24:
yes, the FULLY_CONNECTED crash with reference kernels, but without error info.

njeffrie, 2021-11-24:
Ok, I think I see the issue. TFLM does not support resizing output tensors, but TFLite [does](https://github.com/tensorflow/tensorflow/blob/1fbf82f2031c3a95d7375998edd5870a4bda6a98/tensorflow/lite/kernels/fully_connected.cc#L351). In your model, the output from FC is implicitly flattened to 2-dimensions in the TFLite version of the kernel, but TFLM does not, causing the crash. If you can ensure that the model has 2-dimensional output tensors for the FC operator, the model should work OK.

I am guessing the hifi5 crash is a separate issue. I am less familiar with that code - depending on whether or not you need to minimize CPU cycles, you could possibly disable optimizations to get around the issue, or dig into the guts of the hifi5 implementation.

jiayue198509, 2021-11-24:
ok.Thanks.

closed 2021-11-24 by jiayue198509