tensorflow/tflite-micro #720: Could we omit data type ( int8 or float32) at compile time ?

KarimelQ, 2021-11-23:
I'm using TFLM, and the application only requires kTfLiteInt8 kernels.
However, kTfLiteFloat32 and kTfLiteUInt8 kernels are also built into the application.

It would save a considerable amount of code space if there was a way to disable building in the unused data types ? 

Thank you

advaitjain, 2021-11-23:
This is possible, with some caveats. We are using the term kernel variants to refer to implementations that pull in code for only a subset of the supported types.

Here's an example of how only registering specific types works:
https://github.com/tensorflow/tflite-micro/blob/1a23ad6adf007ad6ef3826523f41ff43fb686a13/tensorflow/lite/micro/benchmarks/keyword_benchmark.cc#L56-L59

Caveats:
  1. The linker must be set up to drop unused symbols from the final binary
  1. The specific variants that you are interested in may not be currently factored out -- we are open to receiving PRs for such refactoring.
  2. Kernel variants have to be implemented in one of the optimized kernel dirs (e.g. cmsis_nn, xtensa etc.)

Here is how FULLY_CONNECTED_INT8 is set up:
https://github.com/tensorflow/tflite-micro/blob/1a23ad6adf007ad6ef3826523f41ff43fb686a13/tensorflow/lite/micro/kernels/fully_connected.h#L63-L81

KarimelQ, 2021-11-24:
Thank you   @advaitjain

advaitjain, 2021-11-29:
Closing this issue. Feel free to open more granular issues if you decide to go down this path.