tensorflow/tflite-micro #721: Build library with Interpreter only dealing with int8

KarimelQ, 2021-11-24:
For memory optimization issues, i'm wondering if there is an arguments to add while building library in order to generate a library with an interpreter for only int8 variables, operations ... 
Thank you

petewarden, 2021-11-24:
We don't currently have support for this, but we don't believe that the interpreter itself would benefit noticeably from specializing on int8, since there's very little code at that level that depends on data types. As @advaitjain comments in #720 this is something we can do for operations themselves though.

Can you share any more details on the current binary size you see, along with the platform you're targeting and build tools? That might help us offer advice on shrinking the footprint.

KarimelQ, 2021-11-30:
Thank you @petewarden for your reply, i'm using MCUxpresso IDE for LPC55S99 board (cortex M-33) , GNU C++11 as compiler, with no optimization, newlibnano library. For binary size, there is more things in the application than TFLM.

closed 2023-06-11 by github-actions[bot]