tensorflow/tflite-micro #722: Build TFLM with CMake

oscarandersson8218, 2021-11-24:
To make it simple for users to integrate TFLM to CMake based projects, such as Mbed-OS (Mbed-CLI 2), we propose adding the option to build TFLM with CMake.

With the make commands:
`make -f tensorflow/lite/micro/tools/make/Makefile list_library_headers`,
`make -f tensorflow/lite/micro/tools/make/Makefile list_third_party_headers`,
`make -f tensorflow/lite/micro/tools/make/Makefile list_library_sources`,
`make -f tensorflow/lite/micro/tools/make/Makefile list_third_party_sources`,
we can list all the needed files to build TFLM. From this it would be possible to generate CMakeLists.txt throughout the build tree.

A simpler implementation of this already exists in the form a [bash script](https://github.com/ARM-software/ML-examples/blob/master/tflm-cmsisnn-mbed-image-recognition/generate_cmake_files.sh). We propose a python script instead as it will work on both Linux and Windows.

petewarden, 2021-11-25:
I couldn't find good documentation on it, but you might want to look at the create_tflm_tree.py script:

https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/tools/project_generation/create_tflm_tree.py

This is now the recommended way to generate these kinds of build files. Is that helpful?

oscarandersson8218, 2021-11-26:
Thanks! I'm handing this over to @mansnils .

mocleiri, 2021-12-11:
@oscarandersson8218 

@vikramdattu  made a cmake file for building tflm here: https://github.com/mocleiri/tflite-micro-esp-examples/blob/master/components/tflite-lib/CMakeLists.txt

line 34 is esp-idf specific but the sources should get you what you need to build.

There is a GNU Makefile example for building the generated tflm here: https://github.com/tensorflow/tflite-micro/tree/main/tensorflow/lite/micro/tools/project_generation

I think you can adapt the tflite-micro-esp-examples  CMakefile to build  then placed into the project_generation directory to provide a quick start for cmake based projects.

Or maybe a general cmake file can be created and then imported by the different platforms to do the final setup of the build.

mansnils, 2022-05-16:
Thanks for your comments. We will park this for now and create a new github issue if there is a need.

closed 2022-05-16 by rockyrhodes