tensorflow/tflite-micro #732: Flatbuffers patch failing

embeddetech, 2021-11-24:
I am running the create_tflm_tree.py script, and it eventually now fails when it is running the patch to Flatbuffers after downloading it. The patch was working some number of weeks ago. The output:

Applying /c/Users/jtork/Desktop/DeleteMe/1/TestTarget/tensorflow_R1K2JZHA/tensorflow/lite/micro/tools/make/downloads/flatbuffers/.//../../flatbuffers.patch to /c/Users/jtork/Desktop/DeleteMe/1/TestTarget/tensorflow_R1K2JZHA/tensorflow/lite/micro/tools/make/downloads/flatbuffers/./
Switched to a new branch 'tflm-patch'
error: patch failed: include/flatbuffers/base.h:1
error: include/flatbuffers/base.h: patch does not apply
error: patch failed: include/flatbuffers/flatbuffers.h:653
error: include/flatbuffers/flatbuffers.h: patch does not apply
error: patch failed: include/flatbuffers/flexbuffers.h:493
error: include/flatbuffers/flexbuffers.h: patch does not apply
tensorflow/lite/micro/tools/make/Makefile:461: *** Something went wrong with the flatbuffers download: .  Stop.

The very last line is the makefile printout because it expects the flatbuffers_download.sh to echo "SUCCESS" and it doesn't. The errors before that are from the patching.

AlexisPapaioannou, 2021-12-08:
Is there any update on it?

humfry, 2022-02-23:
Have the very same issue. Any solution?

closed 2023-06-10 by github-actions[bot]