tensorflow/tflite-micro #740: model run fine without OPTIMIZED_KERNEL_DIR=xtensa in tflm, but model run exception with OPTIMIZED_KERNEL_DIR=xtensa.

jiayue198509, 2021-11-26:
model run fine without OPTIMIZED_KERNEL_DIR=xtensa in tflm, but model run exception with OPTIMIZED_KERNEL_DIR=xtensa.
[model.zip](https://github.com/tensorflow/tflite-micro/files/7607288/model.zip)
[error-log.txt](https://github.com/tensorflow/tflite-micro/files/7607289/error-log.txt)

nyadla-sys, 2021-11-30:
@cad-audio please have a look into this issue.

jiayue198509, 2021-12-01:
XTENSA_TOOLS_VERSION=RI-2020.5-linux
compile cmd：make -f tensorflow/lite/micro/tools/make/Makefile TARGET=xtensa OPTIMIZED_KERNEL_DIR=xtensa TARGET_ARCH=hifi5 XTENSA_CORE=Amlogic_v9 micro_speech_test BUILD_TYPE=release -j4

jiayue198509, 2021-12-02:
[micro_speech_test.zip](https://github.com/tensorflow/tflite-micro/files/7638182/micro_speech_test.zip)
you can use this application to  reproduce the issue.

nyadla-sys, 2021-12-02:
@jiayue198509 this seems to be binary ,could you please share the tflm test application source code that uses your model.

jiayue198509, 2021-12-02:
[src.zip](https://github.com/tensorflow/tflite-micro/files/7638889/src.zip)

jiayue198509, 2021-12-06:
@nyadla-sys have any progress？
If you need any more information, please let me know. Thanks！

nyadla-sys, 2021-12-06:
@jiayue198509 we could reproduce issue from our end and looking into our nnlib-hifi5 that we recently pushed on to Github.

nyadla-sys, 2021-12-14:
we found the issue with our NNLib-hifi5 and working on to fix the issue

jiayue198509, 2022-04-01:
@nyadla-sys have any progress？

nyadla-sys, 2022-04-05:
@cad-audio Bhanu could you please respond to his query?

cad-audio, 2022-04-06:
Hi @jiayue198509,

The issue is related to some alignment in xa_nn_matXvec_sym8sxasym8s_asym8s_circ().
and It is fixed in the latest HiFi 5 NN Library(v1.7.0).
I think tflite-micro master code uses the previous version of HiFi 5 NN Library (i.e. v1.6.0).

Thanks,
Bhanu

closed 2022-04-24 by jiayue198509