tensorflow/tflite-micro #757: input->type is float32 but filter->type is int8

Eddy-Wu, 2021-12-03:
when input->type is float32 and filter->type is int8,  I get the following error:
   tensorflow/lite/micro/kernels/fully_connected.cc Hybrid models are not supported on TFLite Micro.
Is there a way to make this work?
[image]

deqiangc, 2021-12-03:
@advaitjain @njeffrie 

Is this a use case for supplying unquantized data (float32 input) to a quantized model (fixed-point filter)?  This seems strange to me, maybe I am just unfamiliar with this kind of use case. Do you mind elaborating the use case further to help me understand why such a use case is needed?  In case that I misunderstand something, I also add two more teammates in the loop.

advaitjain, 2021-12-03:
TFLM does not support hybrid models (also called dynamic range quantization) so you won't be able to run inference with TFLM for such models.

The reason for this is that with dynamic range quantization the underlying kernel math is still in floating point which is not supported for most of the TFLM optimized kernel implementations. As a result, dynamic range quantized models are not really useful for production use-cases with TFLM.

As such, we recommend using floating point models during prototyping and then switching to full integer quantization when running on your target hardware.

https://www.tensorflow.org/lite/performance/post_training_quantization#optimization_methods

closed 2021-12-03 by advaitjain