tensorflow/tflite-micro #802: micro_speech_lstm_test taking a very long time

advaitjain, 2021-12-17:
The CI time for Xtensa has gone up by a lot (from ~10 mins to 20+ mins). The root cause is the micro_speech_lstm_test.

Running this locally:
```
make -f  tensorflow/lite/micro/tools/make/Makefile TARGET=xtensa TARGET_ARCH=hifi4 XTENSA_CORE=F1_190305_swupgrade OPTIMIZED_KERNEL_DIR=xtensa test_micro_speech_lstm_test -j8
```

confirms that that is the case.

Current workaround is to disable the test.

This slow-down may be related to #590.

tagging @cad-audio

nyadla-sys, 2021-12-17:
@advaitjain  could you please add --turbo in "tensorflow/lite/micro/testing/test_xtensa_binary.sh" file

--- a/tensorflow/lite/micro/testing/test_xtensa_binary.sh
+++ b/tensorflow/lite/micro/testing/test_xtensa_binary.sh
@@ -26,7 +26,7 @@ declare -r MICRO_LOG_PATH=${TEST_TMPDIR}/$1
 declare -r MICRO_LOG_FILENAME=${MICRO_LOG_PATH}/logs.txt
 mkdir -p ${MICRO_LOG_PATH}

-xt-run $1 2>&1 | tee ${MICRO_LOG_FILENAME}
+xt-run --turbo $1 2>&1 | tee ${MICRO_LOG_FILENAME}

 if [[ ${2} != "non_test_binary" ]]
 then

nyadla-sys, 2021-12-17:
@advaitjain  For HiFi4 we have not yet optimized ops, it is using C ref code and hence it is taking more time to run ms_lstm binary

nyadla-sys, 2021-12-17:
@advaitjain we are fine to disable micro_speech_lstm_test example for HiFi4/HiFi3Z core.

closed 2023-06-01 by github-actions[bot]