tensorflow/tflite-micro #807: Bug: duplicate names can cause undefined references in some compilation workflows

cdknorow, 2021-12-20:
There are some files that have the same name in the repo at different paths.

For example: 

tensorflow/lite/micro/kernels/kernel_util.cc 
tensorflow/lite/kernels/kernel_util.cc 

If a user tries to include both of these files in a compiled library, the .a file will contain only the name information and not the path.

ar -t libtensorflow-microlite.a  
   > kernel_util.c
   > kernel_util.c

If at some point in their compilation toolchain, the .a file is converted to object files. One of them will be overwritten and they will end up with undefined references. An example of the error message might be

   _ZTVN6tflite15MicroOpResolverE[_ZT undefined reference to  '__cxa_pure_virtual'

I would recommend renaming any function that is duplicated in the micro folder with a micro prefix

ie tensorflow/lite/micro/kernels/kernel_util.cc  -> tensorflow/lite/micro/kernels/micro_kernel_util.cc 

I would be happy to make a pull request with this fix.

Best

advaitjain, 2021-12-20:
We would be open to a parameter that allows for unique file names in the new project generation script: https://github.com/tensorflow/tflite-micro/blob/a30942eb03efc379e9e80279a44e39078344e5fa/tensorflow/lite/micro/tools/project_generation/create_tflm_tree.py

This would look something like:
```
python tensorflow/lite/micro/tools/project_generation/create_tflm_tree.py --ensure_unique_cc_filenames
```

Happy to discuss some options to enforce unique file names. One option would be to have a flat directory with the file names be the full directory path with `/` replaced with and underscore.

So,
```
tensorflow/lite/micro/kernels/kernel_util.cc 
tensorflow/lite/kernels/kernel_util.cc 
```

would become:
```
tensorflow_lite_micro_kernels_kernel_util.cc 
tensorflow_lite_kernels_kernel_util.cc 
```

The implementation of this functionality would be similar to https://github.com/tensorflow/tflite-micro/pull/704 where we have a stand-alone function that walks through the directory tree and performs the necessary file renaming.

cdknorow, 2022-01-10:
@advaitjain I've been thinking about this for a bit over the holidays. This would most likely address the issue I have but it might be a bit confusing for users in general. Does the tfl_tree parser already rename the imports paths in the files as well?

Perhaps another thing that would work is to have the tf_tree parser check if duplicate file names are being included in the library and raise an error or warning as a note to users. You could also add something to the CI that checks for duplicate file names so you can prevent more of that going forward.

closed 2023-06-01 by github-actions[bot]