tensorflow/tflite-micro #809: Micro interpreter test fails for FVP target

mansnils, 2021-12-21:
Try running 
make -j$(nproc) -f tensorflow/lite/micro/tools/make/Makefile  CO_PROCESSOR=ethos_u OPTIMIZED_KERNEL_DIR=cmsis_nn TARGET=cortex_m_corstone_300 TARGET_ARCH=cortex-m55 test_micro_interpreter_test
It does not work.

deqiangc, 2021-12-21:
@mansnils @advaitjain 

Is this something that works before but stop working lately?  If it used to work, but stop working lately, then I think it may be related with https://github.com/tensorflow/tflite-micro/pull/798

Basically, the change in #898 requires a larger RAM than before and in #798, we have to increase the RAM for certain platforms.  The target that you listed is not in our CI and therefore not caught. 

A couple of options:
1. increase your RAM configuration on your simulation (see #798 for how to do it)
2. I can also try to tuning down the RAM request  a bit.

mansnils, 2021-12-22:
Yes it used to work and yes I think it is related to what you are referring to. My intention was to disable it for now and then fix it later. However I recall you still run this platform as a nightly test, am I wrong?

advaitjain, 2021-12-22:
Yes, this is run as a nightly build: https://github.com/tensorflow/tflite-micro/blob/6b82d6aca5a340f29ee630e4faf3c65112bfaaac/.github/workflows/cortex_m.yml#L52

Disabling helped the build be green and fixing the issue and reenabling sounds good to me.

PR #918 (2022-03-02): Increase stack size for the Arm Corstone-300 target
Increase the stack size for the Arm Corstone-300 target to 0x50000 to fix the micro_interpreter testcase.

BUG=Fixes  #809
PR #1170 (2022-06-11): Change tensor arena memory allocation for Ethos-U Corstone-300 unit tests
The micro interpreter test fails for FVP target. This bug fix changes the allocation of the tensor arena from DCTM to the SRAM memory section. Some other changes were introduced prior to this PR by 'find and replace' operations using sed in the platform.ld file. The sed command is now replaced by applying a git patch on the Ethos-U Corstone 300 platform repository.

BUG=Fixes #809
closed 2022-03-02 by mergify[bot]