tensorflow/tflite-micro #823: Micro-Lite Interpreter

TGiles1998, 2021-12-26:
Is the there a limit to the size of the tflite model that can be interpreted?

mocleiri, 2022-01-02:
As far as I know the main limitation is that the tflm model has to fit into RAM.  Then you also need a bit of space for the tensor arena and also some space for the data collection.  For example dma and application buffers for i2s audio sampling for micro_speech.

This is true whether you are loading a model.tflite from flash or encoding it as a C byte array. 

The larger the model the slower it may be to inference it.  Although it may depend on the hardware acceleration of the ops being performed.

> The OpenMV Cam H7 Plus has about 31MB of frame buffer RAM. That said, running a model anywhere near the that size will be extremely slow.

source: https://docs.openmv.io/library/omv.tf.html

Person detection is the largest reference example at 300,568 bytes for the model.

You should be able to use this filter: https://tfhub.dev/s?deployment-format=lite&tf-version=tf2

to look at tf lite models .  You need to drill down a level and then look at the .tflite model to see the size.

Something like https://tfhub.dev/google/lite-model/object_detection/mobile_object_localizer_v1/1/default/1 which is 1.8 MB
or https://tfhub.dev/agripredict/lite-model/disease-classification/1 which is 11 MB in size.

I think that the Cortex M7 can support > 32MB PSRAM.  The OpenMV Cam H7 Plus has 32 MB (https://openmv.io/products/openmv-cam-h7-plus)

The Teensy 4.1 can support 16 MB PSRAM.
The esp32s3 can support 8MB PSRAM.
The esp32 can support 4MB PSRAM.

Without psram you have a lot less options.

TGiles1998, 2022-01-02:
Thank you very much for the detailed reply!

advaitjain, 2022-01-04:
Thanks @mocleiri for the detailed reply. Cross-linking https://github.com/tensorflow/tflite-micro/issues/822

closed 2022-01-04 by advaitjain