tensorflow/tflite-micro #838: Building hello_world for cortex_m_generic failed with errors

yadhukp, 2022-01-04:
Hi,
I was trying to build hello_world example for my nRF52832 soc so I choose target as cortex_m_generic, but it terminated with errors, what am I missing I'm pretty new to TensorFlow lite micro. I am attaching the logs here
[tfltm_eror.txt](https://github.com/tensorflow/tflite-micro/files/7806159/tfltm_eror.txt)

mansnils, 2022-01-04:
Hi @yadhukp -
the cortex_m_generic target only builds the TFLM-lib. Then you would typically link that into your own application. Your application would then provide those undefined symbols from the log. So with the cortex_m_generic target you provide your own platform support, hence the name generic.

yadhukp, 2022-01-04:
thank you so much @mansnils , now I understand. If it is available, can you suggest any reference material explaining this procedure?

mansnils, 2022-01-04:
You could start by checking out some of the other targets, for example: https://github.com/tensorflow/tflite-micro/tree/main/tensorflow/lite/micro/cortex_m_corstone_300

yadhukp, 2022-01-05:
okay thanks @mansnils

closed 2022-01-05 by yadhukp