tensorflow/tflite-micro #847: DetectionPostprocessFloatRegularNMS hangs for target cortex_m_corstone_300 on Cortex-M55

jenselofsson, 2022-01-10:
The hang occurs for the testcase [DetectionPostprocessFloatRegularNMS](https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/kernels/detection_postprocess.cc#L118) in the [Init-function](https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/kernels/detection_postprocess.cc#L118) when trying to access element at "nms_score_threshold" in the flexbuffers map m.

It also seems to happen when trying to access the following elements in the same map:
nms_iou_threshold
y_scale
x_scale
h_scale
w_scale

The hang occurs in the [ReadScalar-function](https://github.com/google/flatbuffers/blob/master/include/flatbuffers/base.h#L406) in the flatbuffer library, when p is dereferenced.

It happens when KERNEL_OPTIMIZATION_LEVEL is anything other than -O0 for gcc, and a number of optimization flags (-O0, -O1, -O2, -O3, -Ofast) for armclang.

Steps to reproduce:
$ make -f tensorflow/lite/micro/tools/make/Makefile TOOLCHAIN=armclang TARGET=cortex_m_corstone_300 TARGET_ARCH=cortex-m55 test_kernel_detection_postprocess_test

jenselofsson, 2022-01-11:
I seemed to have spoken too soon, I had some local changes laying about which gave a different result. The new results are that GCC works fine, but hangs for armclang for at least -O0, -O1, -O2, -O3.

annietllnd, 2022-06-15:
Hi, here's an update for this issue: Having a look at the hardcoded buffers [g_gen_data_size_non_regular_nms and g_gen_data_size_regular_nms](https://github.com/tensorflow/tflite-micro/blob/89c40b27c6b8ca031012e5b874486988bff00d1a/tensorflow/lite/micro/kernels/detection_postprocess_flexbuffers_generated_data.cc), that are next to each other in memory, we discover that they're char* buffers, with a size of 242 (which is not divisible by 4, the size of a word). The flatbuffer project does not check for alignment. And when the parameter CORE_OPTIMIZATION_LEVEL != -O2, we optimize for size. This means that the second buffer (g_gen_data_size_regular_nms) will be squished and end up over two words, which is believed causes the hang. We will submit a solution where we've added alignas(4) for both of the buffers.

annietllnd, 2022-06-21:
Hi, this issue falls under #274

closed 2022-06-29 by annietllnd