tensorflow/tflite-micro #903: Read only tensor is mapped as input to IF operation

ScholliYT, 2022-02-01:
The below model (when converted to TF-Lite) maps a constant (read only) tensor as input to a `IF` op which causes some problems with the new implementation of the `IF` operation as discussed in #853  with @deqiangc. The expected behavior should be a mapping of the constant tensor directly to the subgraph such that this tensor is not copied multiple times before the execution of the subgraph starts.

The following model tries to implement [slimmable networks](https://github.com/JiahuiYu/slimmable_networks) in TFLM. Instead of universally slimmable Conv2D layers it implements some fixed widths (0.5x and 1.0x) because TFLM doesn't support Dynamic Shapes (see #867). It uses a set of nested `IF` operations to apply `tf.slilce` ops onto the input data and conv2d kernel tensors. 

# Model architecture
```
input: 
- data: float32[1,5,5,10]
- channel_widths: float32[2]

conv1: 
- input: float32[1,5,5,10]
- kernel shape: float32[3,3,10,16]
- output: variable, depending on channel_width but max float32[1,5,5,16]

conv2: 
- input: variable, depending on previous layer output
- kernel shape: float32[3,3,16,32]
- output: variable, depending on channel_width but max float32[1,5,5,32]

MaxPool2D
Flatten
Dense

output: float32[1,10]
```

# How to reproduce
Have a look on the [Colab notebook](https://colab.research.google.com/drive/16guYzn6go7Sps_5XyVQSELMn5Euwlwpb?usp=sharing).
I additionally provide the [tf-lite model](https://drive.google.com/file/d/1BqC9E86y6yONbLrlQqhHdBrAs2zleVUY/view?usp=sharing) and [html file](https://drive.google.com/file/d/1xlM8jJIhbtNKshEaQDMLO85wtBPj-eyy/view?usp=sharing) which can be generated from the notebook above. The tf-lite model can be uploaded to [https://netron.app/](https://netron.app/).

# Visualization
When uploaded to [https://netron.app/](https://netron.app/) we can observe that the `IF` ops have a conv2d kernel with shape `float32[3,3,10,16]` mapped to them. (netron unfortunately doesn't display any subgraphs called by the IF ops).

[image]

Additional info can be found in the html file linked above.

deqiangc, 2022-02-02:
@ScholliYT 

Hello Tom, 

Thank you very much for your effort in creating this detailed description!

Here are my thoughts:

1.  It is quite difficult to address this issue in a general way from the TFLM framework without penalizing more general cases.  When trying to fix the memory overwrite issue in IF, we have thought about other ways such as baking in these kind of constraint in memory plan and avoid copy fundamentally etc, but the complexity is high and some can break some basic assumptions on relationship between subgraphs; so a conclusion is drew that the cost does not warrant more fundamental changes.  We definitely need to watch out if more cases like yours arises.  Currently, there is no immediate plan to support this yet. 
2. To address your problem at hand, one possible faster way is to fix up the tflite model so that the constant tensor is connected with the slice operator directly.  An example that manipulates tflite model is at https://colab.sandbox.google.com/drive/11R9hd_0zvW0y3mesitNtF8dCjf27xP9T#scrollTo=LN0slo1N-dMT.   This does require you to understand the tflite model representation somewhat.  This kind of manipulation is sometime necessary due to the unique constraint in tflm environment which is sometimes different from the slightly more powerful tflite environment. 
3.  A hack is to have your local changes in IF operator to avoid the part of copy back when you detect the memory address is in RO region ( that can probably be done by checking whether pointer is inside the model buffer).  Of course, this is not general and only applicable for local experimental type of changes

ScholliYT, 2022-02-02:
@deqiangc 

Thanks for the quick response. I can totally agree to the points you mentioned. I already adopted a similar local fix like you suggested in 3. which just checks if the copy destination pointer is smaller than some hard coded threshold value (looks like the ro memory region has way lower addresses compared to the normal area). 
I will have a look on approach 2. because this could help with performance in general.

ScholliYT, 2022-02-22:
@deqiangc I think this issue is more appropriate in the main TensorFow repository as the TensorFlow Lite converter causes this behavior. This is not directly related to TFLM. I will move this over to the main repository.

deqiangc, 2022-03-21:
We have since made stride towards memory management. In the latest code, TFLM no longer requires copying back to input. This should solved the original reported issue: copying back to read only tensor. Hence, I am closing this issue.  Let us know if you can confirm this is resolved.  thanks!

closed 2022-03-21 by deqiangc

ScholliYT, 2022-03-21:
Thanks for the update @deqiangc. This is very good news. I will give this a try soon and get back to you.

In the meantime I worked on a solution preventing the copying of read-only tensors altogether by adapting your 2. suggestion from above: manipulating the TF-Lite model. By doing so no buffer memory is required which would only store read-only values anyway. I might make a post on this soon and get in touch with the TF-Lite team. Maybe we can prevent the mapping of read-only tensors to IF operation inputs in advance.

ScholliYT, 2022-07-02:
I just verified and the 3rd hack from above is no longer required. The read-only tensors are not written back from a subgraph anymore. However, the main performance problem of them being copied to the subgraph at all remains.