tensorflow/tflite-micro #907: RNN support for Tensorflow Lite Micro

padoremu, 2022-02-03:


advaitjain, 2022-02-03:
TFLM now supports many of the features (multiple subgraphs, additional OPs) that were blockers when this issue was first created. Please feel free to add any remaining missing features to this issue.

juliusga, 2022-02-10:
@advaitjain Still facing the same issue

StuartIanNaylor, 2022-03-22:
If anyone would like to provide a lstm for the new ESP32-S3 as think xtendsa or is cadence with the LX series have kept that layer behind a paywall.
For me on micro LSTM or GRU was a cul-de-sac for KWS which have some great LSTM/GRU models fave being CRNN as its lite compared to others for very similar accuracy.

Or do I have it wrong and the code does port to ESP32 specifically S3 as that has the vector instructions to make it much more model capable?

advaitjain, 2022-03-23:
This particular issue is talking about RNN with GRU cells. The unimplemented OPs and multiple subgraph support are now part of the TFLM tree. My preference would be to limit this issue for issues related to GRU models.

Please create a separate issue for LSTM support - we are working on that and you can likely expect something in the coming months.

trylaarsdam, 2023-05-16:
> TFLM now supports many of the features (multiple subgraphs, additional OPs) that were blockers when this issue was first created. Please feel free to add any remaining missing features to this issue.

Looks like even though there is LSTM support now in TFLM, the `TensorListFromTensor`, `TensorListReserve`, and `TensorListStack` ops are still not supported.

closed 2023-07-09 by github-actions[bot]

kslavka, 2024-07-10:
Is there a support or tf.keras.layers.GRU in TFlite Micro, the same way it was done for UNIDIRECTIONAL_SEQUENCE_LSTM in 2022. 
Specifically, how to get a single GRU layer in TFLite model instead of unrolled GRU with multiple FullyConected, Mul, SPLIT, and Subs. 
Attaching a [link ](https://medium.com/@christoph-siegl/tensorflow-lite-for-microcontrollers-adds-support-for-efficient-lstm-implementation-25a5f7baa4f6) to an article talking about LSMT support in TFLM.