tensorflow/tflite-micro #911: Support for automatic MicroMutableOpResolver generation

Sebastian-Larsson, 2022-02-04:
<em>Please make sure that this is a feature request. As per our [GitHub Policy](https://github.com/tensorflow/tensorflow/blob/master/ISSUES.md), we only address code/doc bugs, performance issues, feature requests and build/installation issues on GitHub. tag:feature_template</em>

**System information**
- TensorFlow version (you are using): 2.5.0
- Are you willing to contribute it (Yes/No): Yes

**Describe the feature and the current behavior/state.**
Currently, when using an interpreter in TFLM you need to manually add the desired operators, or use an AllOpsResolver. The first option is tedious, and the other has a significantly larger memory footprint. 

We propose an automatic way to generate a MicroMutableOpResolver with only the used operators in a given model. 

**Will this change the current api? How?**
It can be a standalone script that takes a tflite file as input, which doesn't affect the existing API. Alternatively, it can be done at the convert stage, however, this limits the usage, as you cannot get the MicroMutableOpResolver from an existing tflite model.

**Who will benefit with this feature?**
Most TFLM users.

**Any Other info.**

advaitjain, 2022-02-04:
I have moved this issue to the tflite-micro repository. Such a script would be a welcome addition here.

Some additional thoughts and pointers:
 * There are likely some patterns for flatbuffer manipulation and code generation with [Mako](https://www.makotemplates.org/) that can be borrowed from
     https://github.com/tensorflow/tflite-micro/blob/9cd34d1753fd49f374d1a076f6284179f301b338/tensorflow/lite/micro/integration_tests/generate_per_layer_tests.py
    https://github.com/tensorflow/tflite-micro/tree/9cd34d1753fd49f374d1a076f6284179f301b338/tensorflow/lite/micro/integration_tests/templates
 * suggested location for this script is https://github.com/tensorflow/tflite-micro/tree/9cd34d1753fd49f374d1a076f6284179f301b338/tensorflow/lite/micro/tools

 * It would be good to have a test to ensure that this script generates code that can be directly used. Happy to discuss that in more detail once the script is implemented.

tripathidivya, 2022-11-29:
Hello, 

@vamsimanchala @laurenmurphyx64 @advaitjain 
I am new to the tensorflow Lite and I have noticed that many examples are using static tflite::MicroMutableOpResolver <  > micro_op_resolver;. 

So, my questions are:

1. this how many layers we can add here when deploying the model?
2. layers defined under MicroMutableOpResolver  should be exactly same as described in machine learning model?
3. What if there is any layer repeating?

As mentioned here: https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/micro_mutable_op_resolver.h
The Add* functions as mentioned in above link add the various Built-in operators to the MicroMutableOpResolver object. 
So, also as mentioned in this link the layers like convolution and maxpool are 2D layers.

In my project I am using the 1D convolution layer. How can I add Conv1D and Maxpool1D in the main_functions.cc code. This is still not clear for me. 

Can I write my model like This:
	static tflite::MicroMutableOpResolver < 5 > micro_op_resolver; /* NOLINT */
	micro_op_resolver.AddConv1D();
	micro_op_resolver.AddMaxPool1D();
	micro_op_resolver.AddFullyConnected();
	micro_op_resolver.AddSoftmax();
	micro_op_resolver.AddReshape();
	
will that be acceptable?

Could you please explain that to me.

Also, it would be very great if you can share the detailed  link or document for MicroMutableOpResolver ?

Regards,
Divya

mansnils, 2022-11-30:
Hi @tripathidivya 

When using all ops resolver all TFLM operators are included even if they are used or not by the model.
With mutable ops resolver on the other hand you pick only those ops that you want. It doesn't matter if an operator is used many times as long as it is included in the build, so a particular operator is only added once to micro mutable resolver.

If you have a model you can use this script: https://github.com/tensorflow/tflite-micro/tree/main/tensorflow/lite/micro/tools/gen_micro_mutable_op_resolver
It will generate a header file with the needed operators, assuming TFLM have support for that particular model.

If in doubt please use all ops resolver instead. For your project I would recommend all ops resolver, and when that is working you could possibly switch to mutable ops resolver to save some memory.

closed 2022-11-30 by mansnils