tensorflow/tflite-micro #919: Support for ESP32-C3 and ESP32-S3 ?

mtk11, 2022-02-09:
Interested about running tflite-micro on [ESP32-C3](https://www.espressif.com/en/products/socs/esp32-c3) and [ESP32-S3.](https://www.espressif.com/en/products/socs/esp32-s3) Are they supported ? ESP32-S3 is advertised to have AI optimized instructions.

vikramdattu, 2022-02-13:
@mtk11 please find the repo [here](https://github.com/espressif/tflite-micro-esp-examples) with Espressif chipsets support.

This indeed has instructions level optimisations for `ESP32-S3` and general `ANSI C` optimisations in general.

RostakaGmfun, 2022-11-27:
Hi @vikramdattu, do you plan to contribute this port to the upstream?

hadrien-toma, 2023-05-21:
😊 it would still be useful on my side

vikramdattu, 2023-05-22:
@hadrien-toma as I have commented, the support is already available in [this](https://github.com/espressif/tflite-micro-esp-examples) repo.  The repo is periodically kept in sync with tflite-micro. 

What changes having in upstream are useful to you? This isn't planned to be honest. If this indeed looks useful, I can prioritise.

hadrien-toma, 2023-05-22:
Aw sorry @vikramdattu, thank you very much, the repo you mentioned seems to be perfect for my use case! Thanks again ❤️

closed 2023-06-24 by github-actions[bot]