tensorflow/tflite-micro #950: Model with RNN layers fails to load

juliusga, 2022-03-02:
I'm experiencing some issues while trying to use RNN models on Tensorflow Lite. I will try to explain my issue as giving as much detail as possible.

I'm trying out the following simple Tensorflow model, that is using the following data for training:
```
     x_1   x_2        y
...  ...   ...        ...
5    0.0   0.000000   0.055275
6    0.2   0.055275   0.255527
7    0.6   0.255527   1.090631
8    1.1   1.090631   2.069712
9    1.5   2.069712   2.053411
10   1.5   2.053411   2.912017
11   2.5   2.912017   3.855331
12   2.2   3.855331   4.288489
...  ...   ...        ...
```
So I'm preparing the dataframe for the training process, using the following code:
```Python
train_df = train_df[['x_1', 'x_2', 'y']]
train_dataset = train_df.values.astype('float32')
x_train, y_train = train_dataset[:, :2], train_dataset[:, 2]
x_train = np.reshape(x_train, (x_train.shape[0], 1, x_train.shape[1]))
```
This leads me to the following shapes of training features and target:
```
x_train shape: (10753, 1, 2)
y_train shape: (10753,)
```
Then, I'm creating a simple neural network with LSTM layer, using the following code:
``` Python
model = tf.keras.Sequential()
model.add(tf.keras.layers.Input(shape=(1, 2)))
model.add(tf.keras.layers.LSTM(3, time_major=False))
model.add(tf.keras.layers.Flatten())
model.add(tf.keras.layers.Dense(1))
model.compile(optimizer='adam', loss='mae')
model.summary()
model.fit(x_train, y_train, batch_size=32, epochs=10)
```
After training the model, I'm converting it using the TFLiteConverter to use it inside my microcontroller (ESP32).
```Python
import tensorflow_text as tf_text
import sentencepiece as spm

converter = tf.lite.TFLiteConverter.from_keras_model(model)
converter.experimental_new_converter=True
converter.target_spec.supported_ops = [tf.lite.OpsSet.TFLITE_BUILTINS, tf.lite.OpsSet.SELECT_TF_OPS]
converter.allow_custom_ops=False

tflite_model = converter.convert()
open("lstm_model.tflite", "wb").write(tflite_model)
```
This leads me to the following warnings, but the model conversion seems to be successful:
```
WARNING:absl:Found untraced functions such as lstm_cell_layer_call_fn, lstm_cell_layer_call_and_return_conditional_losses while saving (showing 2 of 2). These functions will not be directly callable after loading.
WARNING:absl:Buffer deduplication procedure will be skipped when flatbuffer library is not properly loaded
```
After converting my model, I'm using xxd to make a hexdump:
```console
xxd -i -c 16 lstm_model.tflite > lstm_model.cc
```
After loading the model into the Tensorflow Lite environment, I'm using hello_world example, just to test the load of a model. My current code looks like this:
```C++
...
static tflite::MicroErrorReporter micro_error_reporter;
error_reporter = &micro_error_reporter;
model = tflite::GetModel(lstm_model_tflite);
if (model->version() != TFLITE_SCHEMA_VERSION) {
    TF_LITE_REPORT_ERROR(error_reporter, "Model provided is schema version %d not equal to supported version %d.", model->version(), TFLITE_SCHEMA_VERSION);
    return;
}
static tflite::AllOpsResolver resolver;
interpreter = new tflite::MicroInterpreter(model, resolver, tensor_arena, kTensorArenaSize, error_reporter);
TfLiteStatus allocate_status = interpreter->AllocateTensors();
if (allocate_status != kTfLiteOk) {
    TF_LITE_REPORT_ERROR(error_reporter, "AllocateTensors() failed");
    return;
}
...
```
Everything compiles fine, but when I flash the code into the microcontroller, I get the following error messages:
```
Failed to get registration from op code CUSTOM
AllocateTensors() failed
```
I have tried putting all of the required operators manually, which led me to the following: 
```C++
static tflite::MicroMutableOpResolver<20> *resolver = new tflite::MicroMutableOpResolver<20>();
resolver->AddFullyConnected();
resolver->AddMul();
resolver->AddAdd();
resolver->AddLogistic();
resolver->AddReshape();
resolver->AddQuantize();
resolver->AddDequantize();
resolver->AddShape();
resolver->AddStridedSlice();
resolver->AddTranspose();
resolver->AddPack();
resolver->AddFill();
resolver->AddGather();
interpreter = new tflite::MicroInterpreter(model, *resolver, tensor_arena, kTensorArenaSize, error_reporter);
```
Unfortunately the output was the same. 
I was wondering, is this some kind of a bug, or am I missing something. I have tried various converter parameters, but that didn't make a difference. Is it possible, that I'm missing some operator, that I have to manually add? Maybe there are some ways to list all of the operators, that are being used by the model. 
Thank you in advance for your help and sorry for a long post, I wanted to make things clear!

jokrefta, 2022-04-05:
>  Maybe there are some ways to list all of the operators, that are being used by the model.

You can try using the [model analyzer](https://www.tensorflow.org/lite/guide/model_analyzer) on the model and see what operators it shows being used.

getodo, 2022-05-27:
Did you manage to find a fix? I am have the exact same issue when using a LSTM layer.

juliusga, 2022-05-30:
> 

Unfortunately I did not.

juliusga, 2022-05-30:
> Did you manage to find a fix? I am have the exact same issue when using a LSTM layer.

Unfortunately I did not.

closed 2023-05-24 by github-actions[bot]