tensorflow/tflite-micro #995: LSTM Support

expeon07, 2022-03-25:
Hi, I've been trying to implement a time series prediction model using RNN, but I've been getting an `Only 1 subgraph is currently supported.` error despite following the example from here https://www.tensorflow.org/lite/convert/rnn and [Keras LSTM Example](https://colab.research.google.com/github/tensorflow/tensorflow/blob/master/tensorflow/lite/examples/experimental_new_converter/Keras_LSTM_fusion_Codelab.ipynb#scrollTo=Cf6KC9fbFY5f).

My model is like this:
```
model = Sequential([
    LSTM(5, activation='relu', return_sequences=True, 
            input_shape=(STEPS, FEATURES)),
    Flatten(),
    Dense(FEATURES, activation='relu', name='output')
])
```

Burton2000, 2022-03-30:
You probably want to unroll your LSTM layer before trying to run with TFLu by setting unroll=True. I don't believe there is support for the LSTM fused op in TFLu currently.

expeon07, 2022-04-02:
That worked! Thanks a lot!!

closed 2022-04-02 by expeon07