tensorflow/tflite-micro #1009: Performance regression for conv reference kernel

mansnils, 2022-03-30:
We have seen a performance drop for for the reference kernels, in particular for Arm Cortex-M3 (may be seen in other variants as well).
It seems to be related to https://github.com/tensorflow/tflite-micro/pull/982.

It can be seen with the benchmarks, where there is about a ~7% degradation.

How to reproduce:

make -j -f tensorflow/lite/micro/tools/make/Makefile TARGET=cortex_m_corstone_300 TARGET_ARCH=cortex-m3 run_person_detection_benchmark

Try this on the following sha1: ea94f768500 (after regression) and 645e8ecc7bc (before regression)

rockyrhodes, 2022-05-17:
That commit added group support to the conv2d operator. There is a bit more math in the inner loop keeping track of indices, but on other platforms it did not degrade performance. On the cadence hifi4 platform, I see a speedup of .5-1% with the change, in fact.

Not sure if it's worth your while investigating your compiler results for this example, but I don't think there's anything for us to do in tflite-micro here.

closed 2022-05-17 by rockyrhodes

mansnils, 2022-05-18:
Ok good to know. Thanks.