tensorflow/tflite-micro #1030: Didn't find op for builtin opcode 'REDUCE_PROD' version '1'

argallez, 2022-04-07:
Hi,
I'm trying to convert a transformer network to TFLM. When I try to run my encoder, I get the following error: "Didn't find op for builtin opcode 'REDUCE_PROD' version '1'.An older version of this builtin might be supported. Are you using an old TFLite binary with a newer model? Failed to get registration from op code REDUCE_PROD". 

I am converting the model using tensorflow '2.10.0-dev20220406' (tf-nightly). To incorporate TFLM in my project, I used the files generated by the create_tflm_tree.py script and imported it in DS-5.  My version of the TFLM repo is up to date
The project is then compiled using the Altera Baremetal GCC toolchain.
Here is my model generated by the analyze tool: [encoder_transfo_analyzed.txt](https://github.com/tensorflow/tflite-micro/files/8442451/encoder_transfo_analyzed.txt)
And the flatbuffer verison:  [gradual_mha_custom_28.txt](https://github.com/tensorflow/tflite-micro/files/8442459/gradual_mha_custom_28.txt)

I think this error is linked to the fact that REDUCE_PROD would be available in TFLite but not in TFLM... If this is the case, is there a way to include the op into TFLM? Or would I need to downgrade my version of tensorflow to have an older version of the converter?
Best regards,

AaZz101, 2022-04-11:
Hi, I have the same problem, have you solved it?

argallez, 2022-04-11:
Hi, from what I understand, REDUCE_PROD was not yet included in Micro and these steps should be followed to include it: https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/docs/porting_reference_ops.md . However, I do not currently have the time to follow these myself :/

closed 2022-04-11 by argallez

armando-fandango, 2023-10-17:
has this been implemented in TFLM now ?

rascani, 2023-10-17:
No, REDUCE_PROD has not yet been ported to TFLM. As mentioned in an above comment, the instructions on how to port ops to TFLM are included here: https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/docs/porting_reference_ops.md