tensorflow/tflite-micro #1038: Invoke vailed person_detection on other model

Abdulrahman-CS99, 2022-04-10:
Hello all,

I'm trying to implement my machine learning model which detects a camel instead of a person in the same code of "person_detection".

I changed the model data (the array of Hexa values) and when I uploaded the code to the microcontroller it complied correctly. 

The problem is with the inference stage. The model cannot make any inference and it gave me a static output as shown below:

> Person score: -115 No person score: 72
> Invoke() called after initialization failed
> 
> Invoke failed.
> Person score: -115 No person score: 72
> Invoke() called after initialization failed
> 
> Invoke failed.
> Person score: -115 No person score: 72
> Invoke() called after initialization failed
> 
> Invoke failed.
> Person score: -115 No person score: 72
> 
I am not sure but I think the problem has occurred in these lines of code can anyone help me with this please?
```
  // Run the model on this input and make sure it succeeds.
  if (kTfLiteOk != interpreter->Invoke()) {
    TF_LITE_REPORT_ERROR(error_reporter, "Invoke failed.");
  }
```

rockyrhodes, 2022-05-17:
This indicates a problem initializing the interpreter:

Invoke() called after initialization failed

Possibly a problem allocating enough memory? The log might have further clues as to what went wrong in the lines preceding the ones you've included. If you include more of the output I can better diagnose what is going wrong.

Antoine005, 2023-03-28:
Hi, I'm having this issue and can't solve it here

jcheruvelil, 2024-06-06:
Having this same issue- did anyone end up solving this?