tensorflow/tflite-micro #1045: How to get hexagon_tflm_core.a?

listzhp, 2022-04-13:
I want build a demo which can run on hexagon, so I run test_hexagon.sh in "tensorflow/lite/micro/tools/ci_build/", I am not sure it is a right way, but I got next error:
no such file or directory: '/root/Qualcomm/hexagon_tflm_core.a'.
How to get hexagon_tflm_core.a and what is the of this library?
Thanks.

paulinesho, 2022-04-27:
If you don't need to use the hexagon optimized kernels, you can still build the reference kernels on the hexagon target by bypassing the `OPTIMIZED_KERNEL_DIR`, `OPTIMIZED_KERNEL_DIR_PREFIX`, and `HEXAGON_TFLM_LIB` flags. This means that instead of running `test_hexagon.sh`, run this:
```
make -f tensorflow/lite/micro/tools/make/Makefile TARGET=hexagon
```

listzhp, 2022-04-28:
> 
Thanks for your reply. I want use the hexagon optimized kernels，then how to generate HEXAGON_TFLM_LIB （hexagon_tflm_core.a）？I still confused.

rockyrhodes, 2022-05-16:
hexagon_tflm_core.a is a proprietary library from Qualcomm and not available in the open source TFLM project. I'm not sure of the licensing requirements. You should contact Qualcomm for availability and access requirements.

closed 2022-05-16 by rockyrhodes