tensorflow/tflite-micro #1056: Building on PlatformIO for Teensy 4.0 results in "undefined reference" compile errors.

RationalAsh, 2022-04-19:
Hi! I'm compiling this library for the Teensy 4.0 and the newest version of this library results in a compile error which complains about "undefined references". 

I'm posting this here because an earlier project compiled using an earlier version of this repo compiled perfectly fine (commit commit 219ac1dfed8a8ba0edfdbaae51aed5dc9b208c0c on September 28). Any idea why this might be happening?

Output of the compiling process below: 

> Processing teensy40 (platform: teensy; board: teensy40; framework: arduino)
> -------------------------------------------------------------------------------------------------------------------------------------
> Verbose mode can be enabled via `-v, --verbose` option
> CONFIGURATION: https://docs.platformio.org/page/boards/teensy/teensy40.html
> PLATFORM: Teensy (4.15.0) > Teensy 4.0
> HARDWARE: IMXRT1062 600MHz, 512KB RAM, 1.94MB Flash
> DEBUG: Current (jlink) External (jlink)
> PACKAGES: 
>  - framework-arduinoteensy 1.156.0 (1.56) 
>  - tool-teensy 1.155.0 (1.55) 
>  - toolchain-gccarmnoneeabi 1.50401.190816 (5.4.1)
> LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
> LDF Modes: Finder ~ deep+, Compatibility ~ soft
> Found 98 compatible libraries
> Scanning dependencies...
> Dependency Graph
> |-- <Arduino_TensorFlowLite> 2.4.0-ALPHA+sha.b251e84
> |-- <FlexCAN_T4> 0.0.0+20220416115900.sha.98654c7
> |-- <CircularBuffer> 1.3.3+sha.cf4ded9
> Building in release mode
> Linking .pio/build/teensy40/firmware.elf
> .pio/build/teensy40/src/hello_world.cpp.o: In function `setup':
> hello_world.cpp:(.text.setup+0xea): undefined reference to `tflite::InitializeTarget()'
> .pio/build/teensy40/lib321/libArduino_TensorFlowLite.a(micro_error_reporter.cpp.o): In function `(anonymous namespace)::Log(char const*, std::__va_list)':
> micro_error_reporter.cpp:(.text._ZN12_GLOBAL__N_13LogEPKcSt9__va_list+0x14): undefined reference to `DebugLog'
> micro_error_reporter.cpp:(.text._ZN12_GLOBAL__N_13LogEPKcSt9__va_list+0x1a): undefined reference to `DebugLog'
> .pio/build/teensy40/lib321/libArduino_TensorFlowLite.a(micro_error_reporter.cpp.o): In function `tflite::MicroErrorReporter::Report(char const*, std::__va_list)':
> micro_error_reporter.cpp:(.text._ZN6tflite18MicroErrorReporter6ReportEPKcSt9__va_list+0x14): undefined reference to `DebugLog'
> micro_error_reporter.cpp:(.text._ZN6tflite18MicroErrorReporter6ReportEPKcSt9__va_list+0x1a): undefined reference to `DebugLog'
> collect2: error: ld returned 1 exit status
> *** [.pio/build/teensy40/firmware.elf] Error 1

rockyrhodes, 2022-05-17:
I'm unsure how to replicate this issue. I do see that the Arduino badges are both green on the TFLM page at the moment. Could you try with a version of the code where this is the case and report back?

MATTYGILO, 2023-03-01:
I'm having a simmilar problem with micropython:

```
tensorflow-microlite.c.obj:(.literal+0x18): undefined reference to `TfLiteTypeGetName'
tensorflow-microlite.c.obj: in function `tensor_quantize_float_to_int8':
tensorflow-microlite.c:(.text+0x17b): undefined reference to `TfLiteTypeGetName'
tensorflow-microlite.c.obj: in function `interpreter_get_input_tensor':
tensorflow-microlite.c:(.text+0x1ed): undefined reference to `TfLiteTypeGetName'
tflm/tensorflow/lite/kernels/kernel_util.cpp.obj:(.literal+0x70): undefined reference to `TfLiteIntArrayEqual'
tflm/tensorflow/lite/kernels/kernel_util.cpp.obj: in function `tflite::HaveSameShapes(TfLiteTensor const*, TfLiteTensor const*)':
kernel_util.cpp:(.text+0xc2e): undefined reference to `TfLiteIntArrayEqual'
tflm/tensorflow/lite/micro/memory_helpers.cpp.obj:(.literal+0x4): undefined reference to `tflite::ConvertTensorType(tflite::TensorType, TfLiteType*)'
tflm/tensorflow/lite/micro/memory_helpers.cpp.obj:(.literal+0x28): undefined reference to `TfLiteIntArrayGetSizeInBytes'
tflm/tensorflow/lite/micro/memory_helpers.cpp.obj: in function `tflite::BytesRequiredForTensor(tflite::Tensor const&, unsigned int*, unsigned int*)':
memory_helpers.cpp:(.text+0xce): undefined reference to `tflite::ConvertTensorType(tflite::TensorType, TfLiteType*)'
tflm/tensorflow/lite/micro/memory_helpers.cpp.obj: in function `tflite::AllocateOutputDimensionsFromInput(TfLiteContext*, TfLiteTensor const*, TfLiteTensor const*, TfLiteTensor*)':
memory_helpers.cpp:(.text+0x207): undefined reference to `TfLiteIntArrayGetSizeInBytes'
xtensa-esp32s3-elf/bin/ld: memory_helpers.cpp:(.text+0x233): undefined reference to `TfLiteIntArrayGetSizeInBytes'
tflm/tensorflow/lite/micro/micro_allocator.cpp.obj:(.literal+0x34): undefined reference to `tflite::SingleArenaBufferAllocator::Create(unsigned char*, unsigned int)'
tflm/tensorflow/lite/micro/micro_allocator.cpp.obj:(.literal+0x38): undefined reference to `vtable for tflite::PersistentArenaBufferAllocator'
tflm/tensorflow/lite/micro/micro_allocator.cpp.obj:(.literal+0x3c): undefined reference to `tflite::PersistentArenaBufferAllocator::PersistentArenaBufferAllocator(unsigned char*, unsigned int)'
tflm/tensorflow/lite/micro/micro_allocator.cpp.obj:(.literal+0x40): undefined reference to `tflite::PersistentArenaBufferAllocator::AllocatePersistentBuffer(unsigned int, unsigned int)'
tflm/tensorflow/lite/micro/micro_allocator.cpp.obj:(.literal+0x44): undefined reference to `tflite::PersistentArenaBufferAllocator::~PersistentArenaBufferAllocator()'
tflm/tensorflow/lite/micro/micro_allocator.cpp.obj:(.literal+0x48): undefined reference to `tflite::NonPersistentArenaBufferAllocator::NonPersistentArenaBufferAllocator(unsigned char*, unsigned int)'
tflm/tensorflow/lite/micro/micro_allocator.cpp.obj: in function `tflite::MicroAllocator::AllocateTfLiteEvalTensors(tflite::Model const*, tflite::SubgraphAllocations*)':
micro_allocator.cpp:(.text+0x6b8): undefined reference to `tflite::ConvertTensorType(tflite::TensorType, TfLiteType*)'
tflm/tensorflow/lite/micro/micro_allocator.cpp.obj: in function `tflite::internal::InitializeTfLiteTensorFromFlatbuffer(tflite::IPersistentBufferAllocator*, tflite::INonPersistentBufferAllocator*, bool, tflite::Tensor const&, flatbuffers::Vector<flatbuffers::Offset<tflite::Buffer> > const*, TfLiteTensor*)':
micro_allocator.cpp:(.text+0x778): undefined reference to `tflite::ConvertTensorType(tflite::TensorType, TfLiteType*)'
xtensa-esp32s3-elf/bin/ld: micro_allocator.cpp:(.text+0x8be): undefined reference to `TfLiteIntArrayGetSizeInBytes'
xtensa-esp32s3-elf/bin/ld: micro_allocator.cpp:(.text+0x8dd): undefined reference to `TfLiteIntArrayGetSizeInBytes'
tflm/tensorflow/lite/micro/micro_allocator.cpp.obj: in function `tflite::MicroAllocator::PopulateTfLiteTensorFromFlatbuffer(tflite::Model const*, TfLiteTensor*, int, int, bool)':
micro_allocator.cpp:(.text+0xa5b): undefined reference to `tflite::ConvertTensorType(tflite::TensorType, TfLiteType*)'
tflm/tensorflow/lite/micro/micro_allocator.cpp.obj: in function `tflite::MicroAllocator::MicroAllocator(tflite::SingleArenaBufferAllocator*, tflite::MicroMemoryPlanner*)':
micro_allocator.cpp:(.text+0xb2e): undefined reference to `tflite::SingleArenaBufferAllocator::Create(unsigned char*, unsigned int)'
tflm/tensorflow/lite/micro/micro_allocator.cpp.obj: in function `tflite::MicroAllocator::Create(unsigned char*, unsigned int, tflite::MicroMemoryPlanner*)':
micro_allocator.cpp:(.text+0xb6f): undefined reference to `tflite::SingleArenaBufferAllocator::Create(unsigned char*, unsigned int)'
tflm/tensorflow/lite/micro/micro_allocator.cpp.obj: in function `tflite::MicroAllocator::Create(tflite::SingleArenaBufferAllocator*, tflite::MicroMemoryPlanner*)':
micro_allocator.cpp:(.text+0xbf5): undefined reference to `tflite::PersistentArenaBufferAllocator::PersistentArenaBufferAllocator(unsigned char*, unsigned int)'
xtensa-esp32s3-elf/bin/ld: micro_allocator.cpp:(.text+0xc01): undefined reference to `tflite::PersistentArenaBufferAllocator::AllocatePersistentBuffer(unsigned int, unsigned int)'
xtensa-esp32s3-elf/bin/ld: micro_allocator.cpp:(.text+0xc1c): undefined reference to `tflite::PersistentArenaBufferAllocator::~PersistentArenaBufferAllocator()'
tflm/tensorflow/lite/micro/micro_allocator.cpp.obj: in function `tflite::MicroAllocator::Create(unsigned char*, unsigned int, unsigned char*, unsigned int)':
[... cut ...]
tflm/tensorflow/lite/micro/kernels/l2_pool_2d.cpp.obj: in function `tflite::(anonymous namespace)::L2Eval(TfLiteContext*, TfLiteNode*)':
l2_pool_2d.cpp:(.text+0x500): undefined reference to `TfLiteTypeGetName'
tflm/tensorflow/lite/micro/kernels/l2norm.cpp.obj:l2norm.cpp:(.text+0x114): more undefined references to `TfLiteTypeGetName' follow
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/micropython.elf.dir/build.make:2973: micropython.elf] Error 1
make[1]: *** [CMakeFiles/Makefile2:2068: CMakeFiles/micropython.elf.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
make failed with exit code 2
```