tensorflow/tflite-micro #1062: UNIDIRECTIONAL_SEQUENCE_LSTM run exception in xtensa hifi5 platform.

jiayue198509, 2022-04-20:
UNIDIRECTIONAL_SEQUENCE_LSTM run exception in xtensa hifi5 platform.

error msg: "Number of allocated temp buffers: %d. Checksum passing status: %d"
The possible cause is that dealloc is not called.

The model as follows.
[AI_aec_LSTM-111--1MSE_Synthetic_cus_-12db_int8.zip](https://github.com/tensorflow/tflite-micro/files/8517537/AI_aec_LSTM-111--1MSE_Synthetic_cus_-12db_int8.zip)
.

rockyrhodes, 2022-05-17:
I'm uncertain how to replicate this issue. I do see that the micro_speech_lstm_test currently works @HEAD. Does this work for you as well?

rocky@rocky-ironman:~/tflite-micro$ xt-run tensorflow/lite/micro/tools/make/gen/xtensa_hifi4_default/bin/micro_speech_lstm_test
Testing TestInvoke
Ran successfully

1/1 tests passed
~~~ALL TESTS PASSED~~~

rocky@rocky-ironman:~/tflite-micro$

jiayue198509, 2022-05-17:
You could replicate this issue in hifi5. [xa_nnlib_hifi5_06_30.zip]

root@DESKTOP-P4I3L06:/mnt/d/eric/tensorflow-micro/daily-micro/tflite-micro# xt-run tensorflow/lite/micro/tools/make/gen/xtensa_hifi5_default/bin/micro_speech_lstm_test 
Testing TestInvoke

All temp buffers must be freed before calling ResetTempAllocations()
*WARNING* Unhandled user exception: LoadProhibitedCause (0x00000000)
( Xtensa ISS ) *WARNING* run exited with status 'virtual breakpoint'
root@DESKTOP-P4I3L06:/mnt/d/eric/tensorflow-micro/daily-micro/tflite-micro#

tingyan19, 2022-05-18:
@nyadla-sys can you take a look at this issue? This seems to be specific to the xtensa LSTM implementation.

jiayue198509, 2022-05-26:
@nyadla-sys have any progress on this issue？Please let me know if you have any questions. Thanks.

closed 2023-05-19 by github-actions[bot]