tensorflow/tflite-micro #1102: Wrong prediction every time using the person detection model on ESP32-WROVER-IE board

ASabovic, 2022-05-05:
Hi guys,
I am using the ESP32-WROVER-IE board with the ov2640 camera connected to it. I am trying to run the person detection example and see if I can get some output results based on the captured image. Every time I get same result where the higher value is assigned to not_a_person prediction even if I am present on the picture. I also tried with the existing example but there as well the prediction is always negative (I mean it's always like there is nobody on the picture). Is there any possible solution for this issue? Any kind of help is welcome.

P.S. I am using MicroPython!

vikramdattu, 2022-05-16:
Did you check if you are feeding correct input?
person detection example expects input to be greyscale 96x96. From ov2640, you most likely are obtaining 240x240x3(rgb) image.

ASabovic, 2022-05-16:
In the meantime, I switched to using the Arduino Nano BLE Sense board as the documentation for it is a bit more clear. Now, I am facing another issue as I am not able to connect my camera (OV2640 module) and capture the image in an appropriate way. I am following this tutorial: [https://community.element14.com/challenges-projects/project14/nano-rama/b/blog/posts/tinyml-on-arduino-nano-33-ble-sense---person-detection-with-ble]. Do you maybe know what can cause a problem in this case? From my experience, all Arduino boards should be very easy to use :/

rockyrhodes, 2022-05-16:
Agree that your first issue is probably related to image format. You should try building and running the person_detection_test and verify that gives you the correct result. This uses the two images in tensorflow/lite/micro/examples/person_detection/testdata as positive and negative examples. The input images should be 96x96 greyscale images.

In addition, the GetImage(...) function that needs to be provided by your camera/system should return 8-bit \\signed// integers. This is an unnatural act for most cameras, so you'll probably need a conversion loop somewhere that takes unsigned 8-bit values, subtracts 128, and stores the result in a signed 8-bit value. If you visualize the person.bmp image in testdata/ you'll see that it has already been converted to signed 8-bit values, and hence "looks funny" when viewed with a normal imaging program that will interpret the data as unsigned values.

Hope this helps.

ASabovic, 2022-05-23:
Yes, I first tried with the test images, and there as well I got different results for the different orders of images in my code. For example, if I test first the image with a person on it and then the opposite one, the result will be fine, but if I switch them the result will be completely different. I also thought that problem could be with the size and non-grayscale format of images, but using MicroPython there are no available libraries that can be run on the board directly and convert these images into the right format/size. Based on the last suggestion that I got from the creator of MICROLITE firmware for MicroPython boards the problem can be with the memory and he suspects I am getting memory corruption because the micropython heap and the camera frame buffer are overlapping. By setting the CONFIG_SPIRAM_USE_MALLOC=y flag the heap and frame buffer will be kept separate. But again without the right library for image conversion (the one that already exists for Arduino boards for example), I will again face the issue as the input will not be correct.

closed 2023-05-17 by github-actions[bot]