tensorflow/tflite-micro #1119: Regressions on hotword when using reference kernels

mansnils, 2022-05-11:
FYI. Our CI system caught this in regression. For Arm Cortex-M7 processor there was a performance dip of 9%. For Arm Cortex-M55 processor there was a performance dip of 24%. This was for armclang.

It was introduced with:

commit c8c08497e09cc654065e31c59a6abca6209d48b4 (HEAD)
Author: rockyrhodes <rocky@google.com>
Date: Tue Sep 21 09:58:44 2021 -0700
Add reference kernel support for SVDF op with all 8 bit weights and state (#506)

rockyrhodes, 2022-05-16:
Thanks for the note.  We noticed this on other platforms with this change as well, but weren't too concerned as it just affects the reference implementation. FYI, the cause for the regression is the use of some common code for both 8 and 16-bit versions of the kernel. The (added) int8 version needs to handle asymmetric activation tensors, and the code for handling the potentially non-zero "zero_point" adds a bit of unnecessary overhead to the 16-bit code path, which doesn't support asymmetric activations. We decided the clarity of a single templated version of the code outweighed the performance issue for the case of a reference kernel.

You can see the problem code in svdf_common.cc flagged by this comment:

// The int16 version of the op assumes a zero_point of 0.  This
// code accounts for the potentially non-zero zero_point for the int8
// version of the op.

And yes, I was surprised that this change caused this drastic a performance hit.

closed 2022-05-16 by rockyrhodes

mansnils, 2022-05-17:
Ok. Thanks for the info!