tensorflow/tflite-micro #1157: Didn't find op for builtin opcode 'RESHAPE' version '1'.

GiulioDallaVecchia, 2022-05-25:
Hi,

I'm trying to use tensorflow lite for STM32L4 microcontroller. I'm using STM32CubeIDE to compiler the project. I've generated the Tensorflow tree with:

> python3 tensorflow/lite/micro/tools/project_generation/create_tflm_tree.py --examples hello_world --examples magic_wand --makefile_options="TARGET=cortex_m_generic OPTIMIZED_KERNEL_DIR=cmsis_nn TARGET_ARCH=project_generation" /tmp/tflm-cmsis

After that I train the _magic wand_ model with google coolab, but when I try to use the generated model I get this error:

> Didn't find op for builtin opcode 'RESHAPE' version '1'. An older version of this builtin might be supported. Are you using an old TFLite binary with a newer model?

What is the problem?

Thanks a lot in advance!

Giulio

SherryJingxr, 2022-05-27:
> 

Has your problem been solved? I had the same problem!

SherryJingxr, 2022-05-27:
> > 
> 
> Has your problem been solved? I had the same problem!

> Hi,
>
> I'm trying to use tensorflow lite for STM32L4 microcontroller. I'm using STM32CubeIDE to compiler the project. I've generated the Tensorflow tree with:
>
>
> After that I train the _magic wand_ [...]

Has your problem been solved? I had the same problem!

GiulioDallaVecchia, 2022-05-27:
Hi @SherryJingxr,

Yeap, I've solved the problem. It's just enough to add the op resolver that are missing. 

```
  static tflite::MicroMutableOpResolver<6> micro_op_resolver; /* Remeber to update the number of op */
  micro_op_resolver.AddConv2D();
  micro_op_resolver.AddDepthwiseConv2D();
  micro_op_resolver.AddFullyConnected();
  micro_op_resolver.AddMaxPool2D();
  micro_op_resolver.AddSoftmax();
  micro_op_resolver.AddReshape(); /* I've added this operation */
```
Remember that you have also to update the number of op that you are using (in my case static tflite::MicroMutableOpResolver<6> micro_op_resolver)

closed 2023-05-15 by github-actions[bot]