tensorflow/tflite-micro #1173: Support multi tensor arena

RockySong, 2022-06-01:
Single “tensor arena” is not aware of modern TCM+OCRAM+Ext memory architecture in Cortex-M7, Cortex-M55, Cortex-M85 MCUs, larger model can’t put smaller buffers to faster TCM or OCRAM
Fo
r example, on i.MX RT1170, different memory have very different size and bandwidth:

TCM: Up to 512KB, 8GB/s 
OCRAM: Up to 2MB, 2GB/s
HyperRAM: Up to 64MB, 0.2GB/s read and 0.1GB/s write.

Unlike L1 and L2 caches, they are manually managed, not transparent.

Most models have super large buffers in the early stage but much smaller buffers in the later, if they are put into TCM or OCRAM, then the inference performance can be greatly improved.

jycc1991, 2022-06-21:
> Single “tensor arena” is not aware of modern TCM+OCRAM+Ext memory architecture in Cortex-M7, Cortex-M55, Cortex-M85 MCUs, larger model can’t put smaller buffers to faster TCM or OCRAM
> 
> For example, on i.MX RT1170, different memory have very different size, access power consumption, and bandwidth:
> 
> TCM: Up to 512KB, 8GB/s OCRAM: Up to 2MB, 2GB/s HyperRAM: Up to 64MB, 0.2GB/s read and 0.1GB/s write.
> 
> Unlike L1 and L2 caches, they are manually managed, not transparent.
> 
> Most models have super large buffers in the early stage but much smaller buffers in the later, if they are put into TCM or OCRAM, then the inference performance can be greatly improved.
> 
> As a result, I strongly recommend TFLm to support multiple tensor arenas that with different size, bandwith properties.

Hi
Is it possible to place some part of a  tflm model in system ram and the others in DTCM? 
Thanks.

RockySong, 2022-06-28:
Sure you can manage the linker file to put whatever you want in system RAM. TCM is good for temporarily save intermediate tensors though, and maybe partial of weights when your kernel traverses weights for multiple times to complete an operator; or partial of featuremaps when your kernel traverse featuremaps for multiple times to complete an operator.

closed 2023-05-13 by github-actions[bot]