tensorflow/tflite-micro #1203: Micro interpreter test has invalid link for cloning Arm Ethos-U core platform repo

hmogensen-arm, 2022-06-14:
Current link is not accessible outside of Arm

tingyan19, 2022-06-21:
Cortex-M tests for Corstone 300 (FVP) still fails, please see

https://github.com/tensorflow/tflite-micro/actions/runs/2532980485

These tests failed since https://github.com/tensorflow/tflite-micro/pull/1170 was merged. 

https://github.com/tensorflow/tflite-micro/pull/1204 fixed the link to Arm Ethos-U core platform repository, but the tests still fail with the following error messages:

Applying /home/runner/work/tflite-micro/tflite-micro/tensorflow/lite/micro/tools/make/downloads/ethos_u_core_platform/.//../../increase-stack-size-and-switch-DTCM-SRAM.patch to /home/runner/work/tflite-micro/tflite-micro/tensorflow/lite/micro/tools/make/downloads/ethos_u_core_platform/./
Author identity unknown
*** Please tell me who you are.
Run
  git config --global user.email "you@example.com"
  git config --global user.name "Your Name"
to set your account's default identity.
Omit --global to set the identity only in this repository.
fatal: empty ident name (for <runner@fv-az465-648.l5ewfipqoqhudl1rjdehhrlz2a.jx.internal.cloudapp.net>) not allowed
tensorflow/lite/micro/tools/make/targets/cortex_m_corstone_300_makefile.inc:31: *** Something went wrong with the Ethos-U Core Platform software download: .  Stop.
Error: Process completed with exit code 2.

Maybe call "create_git_repo ./" before apply_patch_to_folder? As in 

https://github.com/tensorflow/tflite-micro/blob/74c6914e58a385c946961f8bc6d0fef189b2c95e/tensorflow/lite/micro/tools/make/pigweed_download.sh#L51-L61

mansnils, 2022-06-22:
Thanks @tingyan19 for spotting this. We will fix.

mansnils, 2022-06-22:
Should be fixed by https://github.com/tensorflow/tflite-micro/pull/1223

closed 2023-05-02 by mansnils